SuperDARN/rst #223: make_fit not writing fields on purpose or a bug?

mts299, 2019-02-28:
Recently I have been testing field checking in pydarn for reading and writing files, and to my surprise, I kept running into records in fitacf files that are missing the fitted fields:
{'x_sd_l', 'x_p_l_e', 'p_l_e', 'x_v_e', 'phi0', 'x_w_l_e', 'nlag', 'w_s', 'v', 'x_gflg', 'w_l_e', 'qflg', 'p_s', 'x_p_s', 'elv_high', 'x_w_s', 'p_s_e', 'x_qflg', 'x_w_s_e', 'gflg', 'sd_phi', 'w_l', 'sd_s', 'phi0_e', 'x_v', 'elv', 'v_e', 'elv_low', 'x_p_s_e', 'x_w_l', 'x_p_l', 'w_s_e', 'sd_l', 'p_l', 'x_sd_s', 'x_sd_phi', 'slist'}
At first I thought I found a bug but soon realized that this happened in the other fitacf files I produced so now I am thinking these fields are not written when there is "bad data" however the record scalar information is still written in. 

@asreimer told me lmfit2 would not do this because it fits everything. So now I see inconsistencies in the code and file generation. 

I am wondering if we need to review this?  
@egthomas @asreimer @pasha-ponomarenko thoughts or explanations?

egthomas, 2019-02-28:
Certain fields are only included if the XCF data is present in the rawacf file (eg elevation angles, everything beginning with `x_`, etc).  I'm guessing the other fields may not be written due to bad quality flags.  Here's what an old [tutorial](https://github.com/SuperDARN/rst/wiki/Using-The-IDL-DataMap-Library-With-New-Format-Files) had to say:

> The problem with fitacf files is that depending on whether a good fit was obtained, a range may not be stored in the file. The array variable slist contains a list of the ranges that were stored.

mts299, 2019-02-28:
@egthomas you are the librarian of RST documentation. 

Moving forward, is this something we might want to change to handle it a bit more elegantly? Otherwise people like myself might think it is a bug or think something is wrong with the data. It can be a bit misleading.

asreimer, 2019-03-04:
Exactly what @egthomas said. This is only the case for fitacf2.5 though. Here's the specific references in the code that explains why this happens:

`make_fit` calls `FitACF` and then outputs the results using `FitFwrite` in a while loop (see [here](https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.bin/tk/tool/make_fit.1.19/make_fit.c#L301))

`FitACF` calls `do_fit` [here](https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.lib/tk/fitacf.2.5/src/fitacf.c#L194) and in `do_fit`, each range is fitted one at a time, with the result of the fit setting a quality flag [here](https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.lib/tk/fitacf.2.5/src/do_fit.c#L235-L239). If the quality flag is 1, a bunch of parameters get set (`p_l`, `p_s`, etc), if not, they don't. This happens for both the ACF and the XCF. When parameters aren't set, they remain at their initialized value: 0 (except SNR, which is -50).

`FitFwrite` calls `FitWrite` ([here](https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.lib/tk/fit.1.35/src/fitwrite.c#L249-L270)), which calls several different functions, but for your question, `FitEncode` is the most important one, specifically [this line](https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.lib/tk/fit.1.35/src/fitwrite.c#L185) which prevents any arrays from being written if the quality flag is not 1.

Whether this is a problem or not really depends on your perspective. Looking at the whole process of fitting data from a statistical point of view, setting any kind of quality is a subjective choice, so by doing so you are imposing your subjective choice on any user downstream that might have different and perfectly valid justifications for a different subjective choice. Saving the chi-squared and fitted parameter errors allows users to decide what quality of data is good enough for their study. Historically the choice seems to have been made to save storage space, which is really not a problem any more.

asreimer, 2019-03-04:
@mts299 Addressing this might fall under the banner of "modifications to fitacf2.5", since it fundamentally changes the contents of a fitacf file in a way that users are not used to. So probably not something that can be addressed in fitacf2.5.

Looking at the code, I echo @egthomas previous comments about how difficult it is to make changes to the existing code (elevation library) due to the build system and how fitacf2.5 and fitacf3.0 are inter-mingled. Addressing this issue (arrays not being written if `qflg!=1`) for fitacf3.0 could be done by setting the quality flag to 1 no matter what (kind of hacky...) or, a large rewrite of make_fit and other libraries would be needed (much better, but a lot more work).

mts299, 2019-03-04:
@asreimer I still think it is an issue in Fitacf 3.0 
From a fitacf 3.0 file 
        int     "fitacf.revision.major" = 3
        int     "fitacf.revision.minor" = 0
        float   "noise.sky" = 15.382937
        float   "noise.lag0" = 0.000000
        float   "noise.vel" = 0.000000
arrays:
        short   "ptab" [7]
        short   "ltab" [19][2]
        float   "pwr0" [100]
scalars:
        char    "radar.revision.major" = 1
        char    "radar.revision.minor" = 18
        char    "origin.code" = 0
I think lmfit2 doesn't do this?

asreimer, 2019-03-04:
@mts299 you are correct. lmfit2 doesn't do this because it should be up to the user to decide what data is "good" or "bad" for their study. I think early on in the development of fitacf3 Keith was keeping all the ranges and maybe that's what I was remembering. Sorry about the confusion.

mts299, 2019-03-04:
I think there is a couple of options on how to solve this issue if we wish to address it:
1. Do nothing :p as in we do not address it 
2. drop records if the data is not usable - again might raise some eyebrows on why records are gone but it keeps with consistency. Also this may require the least amount of code changes. 
3. have a bad data flag field in the record and still do fitting - if this is mathematically possible also will require more changes to other procedures. 

3. supports the consistency practice and allows users to choose what they want to do with the data. However, I don't know what mathematical limitations are imposed on the fitacf fitting procedures ¯\_(ツ)_/¯

mts299, 2019-06-13:
As a follow up to the workshop on this issue, we still need to figure out what determines the `qflg` in the code which is not present in RAWACF file and does not get written to in the FITACF file if `qflg != 1`. 

I think the `qflg` in the code vs. in the file got the PI's confused. Also, I looked through a FITACF file quickly and did not see a single instance where `qflg` was `!= 1` so it doesn't look like we are even using this field effectively. 

I will dig deeper into the code when I have some time to try and get a full understanding of what is going on, but it seems we may a have couple more problems on our hands :/

egthomas, 2019-06-13:
`qflg` and `xqflg` are set by the return value of [fit_acf](https://github.com/SuperDARN/rst/blob/develop/codebase/superdarn/src.lib/tk/fitacf.2.5/src/fit_acf.c) in [do_acf.c](https://github.com/SuperDARN/rst/blob/develop/codebase/superdarn/src.lib/tk/fitacf.2.5/src/do_fit.c) of FITACF 2.5.  If the fitting process succeeds then they are set to 1, otherwise they are set to -1, 2, or 4 depending on the reason for failure.  If there was not a successful fit then there is no velocity, power, width, etc information stored in the `FitRange` structure.  `slist` in the final `fit` or `fitacf`-format file is the list of ranges where qflag was equal to 1.

Some of our confusion stemmed from the fact that when a `fitacf` file is opened in IDL, a structure with the full `nrang` number of elements is created, but only those with `qflag=1` are actually populated with values.  This stems from the [FitSetRng](https://github.com/SuperDARN/rst/blob/develop/codebase/superdarn/src.lib/tk/fit.1.35/src/fit.c#L63) function which is typically used in IDL via the DLMs.

So I'm not quite sure what the issue is here, other than it not being immediately obvious that the full collection of fitted parameters is not stored for all ranges in a `fitacf` file for FITACF 2.5 or 3.0.

egthomas, 2019-06-13:
apologies to @asreimer, who already wrote a much more thorough explanation several months ago.  My basic feeling that this is a non-issue with respect to FITACF 2.5 or 3.0 still stands.

mts299, 2019-06-13:
If there is a failure in `fit_acf` or `do_acf` then the `qflg` will never be written, correct? 
If it is successful then the only value written for `qflg` will be 1, correct?
What are the conditions that set `qflg!=1`? 

@egthomas  if this is not an issue, what should we do to let others know why some fitacf records are missing fitted fields? What if they wanted access to "bad" data?

DannoPeters, 2019-06-14:
While processing a large set of fitacf files, my code threw repeated key errors. I consulted the fitacf documentation (https://superdarn.github.io/rst/superdarn/src.doc/rfc/0009.html) and found nothing regarding this issue, and that my code was parsing the fitacf files correctly. I was initially concerned that the field names in fitacf documentation may have been wrong; upon closer inspection of the fitacf files using dmapdump, I saw half missing records and was quite concerned that the fitacf files were corrupted.  

I tried re-downloading the fitacf files and testing with many different files, seeing that almost every file threw errors, I thought that a large portion of the SuperDARN dataset might have been corrupt. 

Now seeing that the partial records are by design, is there a cleaner way of marking "bad" data? 
Would it be more efficient to either not include the records with omitted data fields, include null data fields, or include the "bad data" and add a field to flag data considered poor quality?

ecbland, 2019-06-14:
As others have mentioned already, the "missing" XCF fields is the intended behaviour when there is no XCF data. You may have noticed similar behaviour when using `make_grid -xtd`, which creates additional fields in the output file. If the only issue here is **confusion** then it can be addressed in the documentation (e.g. in the file format description). 
@mts299 : are you already working on the FITACF file format description for the RST readthedocs? If not, I can do it. 

Whether or not the `qflg!=1` data should be written to file is a separate issue relating to the file format and it warrants further discussion with input from the PIs. I'm going to start keeping a list of longer-term issues like this so we can discuss them in a more structured manner.

mts299, 2019-06-14:
@ecbland I haven't had time to work on the documentation as of so far for RST due to other commitments. But it is on my next todo, I have already jotted down a lot of notes on where the documentation is outdated with the current FITACF, RAWACF, and IQDAT file format. 

In my opinion, mentioning how a command line option that adds fields is easier to document and for researchers to understand as it should propagate through the whole file. Then explaining how we have a specific method that takes out the fitted fields that make up a FITACF record at random points in the FITACF file, I think the loss of consistency in the format is confusing part. Either way, some documentation is better than none :p 

I wanted to play around with `make_fit` code to see how much of an impact there would be by adding back the data and utilizing the `qflg`. I think when we brought this up to the PI's at the workshop, there was some confusion around this issue, so having a potential solution or an understanding of the workload needed may aid in picking a resolution. 

My main concern is how many people already think our data is corrupt, and how often will they know to look into the documentation to find the explanation? 

@ecbland I think this issue is regarding if we should add the data back :) At least that was my intention when making this issue.  

I can make another issue to address the documentation portion.

ecbland, 2019-06-15:
Ok, @mts299 is addressing the confusion aspects of this issue by updating the documentation (issue #230). Thank you :)

@mts299 : 

> I think this issue is regarding if we should add the data back :) At least that was my intention when making this issue.

Including the `qlfg!=1` data in the FITACF files is a scientific decision which should be driven by the wishes of the community. In my view, making this decision goes beyond the scope of this WG, and we should consider addressing it only if there is a specific request to do so with approval from the PIs. That request can, of course, come from people within the DAWG (and I would support it). I am aware that there was some discussion on this topic at the workshop but there didn't appear to be a conclusion. Note that including the "bad" fitted data creates additional potential for confusion (or worse, misinterpretation), so this decision requires careful consideration.

As co-chair of this WG, I will follow this up with the PIs. We can also discuss it during our next skype meeting. But in the interests of focusing our time & energy on issues we can resolve, **are there any objections to closing this issue for now?**

asreimer, 2019-06-16:
@ecbland 
> Whether or not the qflg!=1 data should be written to file is a separate issue relating to the file format and it warrants further discussion with input from the PIs. I'm going to start keeping a list of longer-term issues like this so we can discuss them in a more structured manner.

Sort of, but it's really not a file format issue. This issue is much more of a question of data quality and data distribution and whether or not the DAWG should be imposing subjective filtering (I list them below) on the data at all and if that filtering is objectively justifiable. So I think that the issue is 2-fold.

## Issues
1) the FITACF file format description doesn't accurately reflect what is actually written to the FITACF files (what does `make_fit` output?), and
2) What is written to the FITACF files and why is not objectively justifiable. For some historical/ad hoc/empirical/file sizes reasons, it was decided long ago not to write data to the FITACF files where `qflg!=1`, which kind of defeats the whole point of having the qflg in the files in the first place. Additionally, what determines `qflg!=1` is entirely subjective (see below for examples).

Issue 1 is probably best solved with documentation (#230).  Issue 2 requires more work and is a bit more complicated than the current "what do we do about qflg" problem.

In my opinion, the decision to include all data in FITACF files (both `qflg!=1` and `qflg=1`) is well within the authority of DAWG and the best decision. And the criteria determining the qflg value should be improved. There are simple and well established (see numeric recipes) ways to determine whether a fit is "good" and if the resulting fitted parameters are "meaningful" and these involve goodness of fit tests and construction of confidence intervals and the qflg should probably be based on those criteria. We don't really do this properly in FITACF2 (although things have improved with FITACF3 and LMFIT2!) and it ultimately involves a subjective choice that others might not agree with. So DAWG can package a qflg, but should stop subjectively filtering the data where `qflg!=1`. Instead, give the users the info they need and let them decide. DAWG can include a qflg based on well-documented criteria, but we shouldn't prevent others from imposing their own subjective choice by not saving all of the fitted data!

## Potential Solution
In my opinion, the best thing to do is for DAWG to come up with an updated data processing plan and then take that to the PIs. There's already been many discussions about what to do, so maybe we have enough ideas now to propose something. To that end, here's a very rough sketch of what I see that might work:

1) Adopt a "data level system" to define data products produced by SuperDARN, similar to the NASA data level system:
    a. level 0 is IQ data
    b. level 1 is ACF data
    c. level 2 is fitted parameters derived from ACF data
    d. level 3 is derived data products (convection maps, field aligned currents, geolocated fitted data, index of refraction corrected fitted data, etc)

and within each of these levels there can be sub-levels, such as level 2b: fitted parameters with data quality flags and metadata available. level 2 would be dmap files and level 2b could be hdf5 files that have all information needed for a scientist to do science without needing RST specifically. DMSP data provided by UT Dallas has adopted a 1, 2, 3, 4 data quality flag system ([this paper describes the flags](https://www.ann-geophys.net/23/2479/2005/)). Why do we need to only use qflg = 0 or 1?

2) Stop imposing our own (the DAWG's) subjective choices on the data. If the data files contain enough information (metadata) and the processing is done properly (proper error bar calculations, chi-squared, fitcodes from the Levenburg-Marquardt are recorded, etc), then the science user can work with fitted data files and select the appropriate error thresholds that are appropriate for their own study. This makes the DAWG's job easier long term and gives scientists the flexibility that they need/want. In my experience, most users of SuperDARN data already impose their own data filtering/rejection schemes in addition to the existing `qflg=1` criteria.

The only downside to this is that file sizes will increase, but the benefits outweigh this, especially since files can be compressed such that file size increase is marginal.

I understand there will be some kind data task force or something like that. Maybe this is a perfect task for them? I think the DAWG can easily resolve this issue. Maybe it's too late for FITACF2, but going forward, things can be improved with minimal effort.

## Apendix S
**Subjective Filtering**
Note that I've only looked at FITACF2 here since this is the data most people are working with. FITACF3 and LMFIT2 are better about this, although [FITACF3 is still subjectively throwing out data ](https://github.com/SuperDARN/rst/issues/223#issuecomment-469444496).

1) [Just because SNR < 1 doesn't mean the fitted parameters are meaningless](https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.lib/tk/fitacf.2.5/src/fit_acf.c#L74-L76). To first order, the relative uncertainty in the lags of an ACF is (1+1/SNR)/sqrt(K), where K is the number of samples. So using 30 pulse sequences and SNR=1, that's an uncertainty of 36%. But the fitting procedure combines several lags together so the fitted uncertainty isn't necessarily too large to be useful!
2) [Similar problem here](https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.lib/tk/fitacf.2.5/src/fit_acf.c#L159-L163). This is a "filtering by the uncertainty in the lag 0 estimate" and is equally subjective and not always valid.
3) [Yikes...](https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.lib/tk/fitacf.2.5/src/fit_acf.c#L196-L213) This throws away data such that it's possible to only end up fitting 3 lags.

ecbland, 2019-06-16:
@asreimer I completely agree with you, and I would really like to see the subjectivity removed from FITACF2.5/FITACF3.0. 

If I appear to be pushing back on everything here, it's because this discussion has grown much bigger than it started out. I read the minutes from the DAWG meeting on 4th June: 

> Issue 223: make_fit not writing fields on purpose or a bug?
Description: During fitting of RAWACF files, if the data are not of good quality or fitting
process is not converging, the parameter fields/arrays are not recorded. Specific example: ACF
data are recorded but XCF data are not. What should be done about this?
Conclusions/Solution: Run further research into what criteria are used to drop the fitted fields
and produce better documentation on this process to clarify what the quality flag actually means. 

There is no mention here of changing FITACF2.5/FITACF3.0 to actually write the `qflg!=1` data into the files. Only understanding/documenting how the data are filtered. 

It takes a lot of time/energy for all of us to follow these online conversations, especially from different time zones. This is a really important issue and I want to see it implemented. I just think we need to discuss it together on skype to decide how to proceed.  @mts299  has also alerted me to some broader issues relating to data policies which are being discussed at USask with Kathryn, so some input from that discussion would be useful here. So I suggest we put this issue on the agenda for our next DAWG skype meeting and pause the online discussion. We can then focus our energy on issues/PRs relating to the next RST release.

mts299, 2019-06-16:
As a middle ground solution, because I don't know when the DAWG skype call will be. Could we add an option in `make_fit` to include all fields and use the `qflg` then maybe people can test it or see the differences? This may help with understanding the problem more and either to present to others and PI's? 

I think right now we are unsure of the impact it will have/code changes to further products :) A good test of theory and I don't think the change will be too hard thanks to @asreimer pointing out where it doesn't write the fields in :)

asreimer, 2019-06-16:
I mostly agree, where I disagree is with talking to the PIs about this and other small issues. The DAWG exists so the PIs don't have to do this stuff.

I don't think the qflg discussion here is irrelevant or out of scope though. The fundamental reason that the data is missing is because of the qflg filtering (with the exception of the XCF arrays which also depend on xcf existing). The June 4th meeting minutes seem to align with this discussion nicely? 

In any case, there does seem to have a consensus in the DAWG about what needs to be done, but not really a concrete plan to get it done. Now seems like a good opportunity to make such a plan and get it done.

I'm also very sensitive to the fact that everyone here is volunteering time that they probably don't have much of to begin with.

Definitely this should be talked about in a meeting specifically dedicated to it.

asreimer, 2019-06-16:
@mts299 
> I think right now we are unsure of the impact it will have/code changes to further products

Ya that's why I think it might be too late for FITACF2. But this could be done for 3!

ecbland, 2019-06-16:
@mts299 
> Could we add an option in make_fit to include all fields and use the qflg then maybe people can test it or see the differences? This may help with understanding the problem more and either to present to others and PI's?

Yes, good suggestion.

pasha-ponomarenko, 2019-06-17:
I agree with Marina that it is necessary to keep in mind what happens downsteam, i.e., to make sure that the low-quality data from *.fitacf files are not gobbled uncritically by gridding/mapping/E-filed routines.
Yet another thing: it is a good practice to approach the PIs with a solution or two rather than with a question or two.
;-)

sshepherd, 2019-06-17:
Hi Emma,

You were missed at the workshop, but I am glad to see that you are
remaining involved in the RST. As an observer and former contributor I am
wondering if there are any changes in the charter, membership, rules of
engagement, etc. that we all (or maybe just me) should be aware of. I will
add there there are at least a few PIs who are interested in our software
and data file formats, and changes that are being suggested. Many groups
have very limited resources and changes to underlying formats could have a
big (and negative) impact on scientific productivity within the group (or
individual.)

Thanks for agreeing to take on this task!

Best,
Simon

ecbland, 2019-06-17:
@sshepherd  No changes to the WG charter/rules etc. But as you said, there are potential negative consequences to implementing the changes that have been suggested, so I think it warrants discussion with the PIs before we invest time implementing them. Of course, we need to do some initial testing to see what is possible and what the proposed changes might mean for data users (@mts299 is doing this).

ecbland, 2019-06-17:
Closing this issue with the following partial resolution:
1. The documentation on the FITACF file format will be updated (issue #230)
2. The quality flag definitions will be determined and documented (issue #230)
3. @mts299 is doing an initial investigation into how to include the `qflg!=1` data from FITACF2.5 and/or FITACF3.0. Let's wait for her to report back in a new issue/PR. 

The remainder of this discussion is important but too broad to be productive here. It will be addressed during the next telecon.

closed 2019-06-17 by ecbland