SuperDARN/rst #224: How to modify the content in the "rst/.profile/idl.bash"?

Guojun-li, 2019-03-14:
I have installed the IDL8.3 in the directory of "/usr/local/exelis/", and the IDL_IPATH in the "rst/.profile/idl.bash" is modified. Here is the modified content in the "rst/.profile/idl.bash":
# idl.bash
# ========
# IDL header directory
export IDL_IPATH="/usr/local/exelis/"
# RST DLM directory
export DLMPATH="${RSTPATH}/dlm"
# IDL environment
export IDL_PATH="<IDL_DEFAULT>:+/${RSTPATH}/idl/lib"
export IDL_DLM_PATH="<IDL_DEFAULT>:${RSTPATH}/dlm"
export IDL_STARTUP="${RSTPATH}/idl/startup.pro"
# Makefile prototype for IDL DLMs
export MAKEDLM=${BUILD}/make/makedlm

But when I run the IDL, there was an error:
% Error opening file.
  File: /home/zm/Folder_lgj/download/rst/idl/startup.pro

Then I modified the  content in the "rst/.profile/idl.bash"  as bellow:
# idl.bash
# ========
# IDL header directory
export IDL_IPATH="/usr/local/exelis/"
# RST DLM directory
export DLMPATH="${RSTPATH}/dlm"
# IDL environment
export IDL_PATH="<IDL_DEFAULT>:+/${RSTPATH}/idl/lib"
export IDL_DLM_PATH="<IDL_DEFAULT>:${RSTPATH}/dlm"
export IDL_STARTUP="${RSTPATH}/codebase/superdarn/src.idl/startup/startup.pro"
# Makefile prototype for IDL DLMs
export MAKEDLM=${BUILD}/make/makedlm

And I got these errors when I run the IDL:
Error opening file. File: genmag
% Error opening file. File: astalg
% Error opening file. File: igrflib_v2
% Error opening file. File: aacgmlib_v2
% Error opening file. File: aacgm_v2
% Error opening file. File: time
% Error opening file. File: mlt_v2
% Error opening file. File: datamap.pro
% Error opening file. File: aacgmdefault.pro
% Error opening file. File: aacgm.pro
% Error opening file. File: radar.pro
% Error opening file. File: rprm.pro
% Error opening file. File: iq.pro
% Error opening file. File: raw.pro
% Error opening file. File: fit.pro
% Error opening file. File: grd.pro
% Error opening file. File: cnvmap.pro
% Error opening file. File: fitcnx.pro
% Error opening file. File: cfit.pro
% Error opening file. File: oldraw.pro
% Error opening file. File: oldfit.pro
% Error opening file. File: oldgrd.pro
% Error opening file. File: oldcnvmap.pro
% Error opening file. File: fitacf.pro
% Error opening file. File: efield.pro
% Error opening file. File: genlib.pro
% Error opening file. File: acflib.pro
% Error opening file. File: rawlib.pro
% Error opening file. File: gridlib.pro
% Error opening file. File: maplib.pro
% Error opening file. File: fitlib.pro
% Error opening file. File: aacgmlib.pro
% Error opening file. File: igrflib.pro
% Error opening file. File: geoplib.pro
% Error opening file. File: rbposlib.pro
% Error opening file. File: istplib.pro
% Error opening file. File: global.pro

What is wrong with the content?

egthomas, 2019-03-14:
@Guojun-li thanks for the question.  Let me start by asking a couple of additional questions:

1) Do you know which version of the RST you have installed?
2) Is the `idl_export.h` file located within the `/usr/local/exelis` directory on your machine?  Or is it under some kind of `include` subdirectory?  If so, the `IDL_IPATH` environment variable needs to point to the specific directory where `idl_export.h` is located.

The first issue you identified (`% Error opening file.  File: /home/zm/Folder_lgj/download/rst/idl/startup.pro`) leads me to believe that when you first compiled the RST, it could not find the `idl_export.h` file in the `IDL_IPATH` directory, so it skipped over installing the IDL DLMs and creating the symbolic links to the IDL code.  You can confirm this by seeing if there is an `idl.tar.gz` archive within the `codebase/superdarn/src.lib/tk` directory.  

Please try updating your `IDL_IPATH` and then recompile the RST with `make.code` and let us know if you see any additional errors.

Guojun-li, 2019-03-14:
@egthomas thanks for your reply. 
1.I do not know the version of the RST I have installed, but I downloaded the rst-master.zip from the github (https://github.com/SuperDARN/rst) about 10 days ago.
2.The idl_export.h is under the subdirectory named /usr/local/exelis/idl/external/include.

Thank you for your help.I update the IDL_IPATH to make it point to /usr/local/exelis/idl/external/include, and recompile RST, and then there is no errors.

egthomas, 2019-03-15:
@Guojun-li no problem, happy to help.  Unless you have any other questions or concerns I'll go ahead and close this issue.

closed 2019-03-16 by egthomas