SuperDARN/rst #230: Inconsistency in FITACF file format Documentation

mts299, 2019-06-14:
This issue is to address that our documentation on FITACF file format needs to be updated to explain why and how some FITACF records are only partially written due to "bad" data.

Other comments on what is missing and needs to be updated in the FITACF file documentation can also be placed here. 

Documentation todos
- [ ] correct data types 
- [ ] Warning on FITACF file format inconsistency
- [ ] correcting units in some fields 
- [ ] adding units to other fields 
- [ ] further information on `qflg` current state 
- [ ] further information on `x_qflg` int meanings

egthomas, 2019-06-17:
@mts299 which particular documentation file(s) are you suggesting need to be updated?

mts299, 2019-06-17:
I will be updating the information in readthedocs as @ecbland mentioned.

asreimer, 2019-06-17:
@egthomas see discussion in #223

egthomas, 2019-06-17:
So from reading the '19 DAWG meeting minutes, I've noticed that some of the brief conclusions/solutions listed are not representative of the discussions which took place during the WG meeting.  There were a lot of concerns raised by several attendees (PIs and WG members alike) about wholly stripping out documentation from the distributed RST package and moving it to an external platform.  The solution offered that evening was to build the information on the readthedocs page from the XML documentation files already present in the RST - hence my question about where the existing documentation was going to be updated.

asreimer, 2019-06-17:
@egthomas I don't think anyone here is suggesting that we strip out documentation from RST. I think this is a wild misconception/misunderstanding.

There is nothing wrong with having **BOTH** a read the docs page AND the XML documentation. Users can use one, or both, of these documentation options.

So, update the XML documentation, specifically pertaining to how FITACF fields are written depending on qflg (as discussed in #223) and others can add this to the read the docs page too.

asreimer, 2019-06-17:
@egthomas can you point out when/who ever suggested we would remove the existing documentation in RST? This is the first I've ever heard/read about this concern and it seems to only be you with the concern?

sshepherd, 2019-06-17:
I have certainly had the impression that this was the intent. I also think
your suggestion is somewhat flawed, but perhaps I don't understand it. You
seem to be suggesting that two separate sets of documentation are updated
(or not) independently from each other.
Is that the case? If so, I think that is a terrible idea and definitely
*NOT* what was stated at the WG group meeting in Japan.

asreimer, 2019-06-17:
Maybe this happened after you turned off email notifications, but the discussion in #210 is informative. The only talk of "transfer of information" was in the context of installation instructions and code example usages. Specifically, the purpose of the read the docs page was to host the RST for dummy's guide.

I think it's good for us to all understand that we shouldn't strip out the XML documentation, but there's no plan to do that and no one is currently doing it.

mts299, 2019-06-17:
I can remove the `make_fit` and `make_grid` links in the table of contents if that is where the confusion is coming from? I put it there as the thought of maybe including it there but I won't do that now.

The XML documentation and command line help is another issue and was mainly brought up by @egthomas. As mentioned in the teleconference the readthedocs is a nice central place for information on installation, howto's and DMAP format information, which was agreed upon as a good idea from the majority of people there. 

The XML, API stuff can wait. That is another issue and out of the scope of this issue.

ecbland, 2019-06-17:
Just to reiterate what @mts299 said above: **please restrict this discussion to the specific topics raised in the issue description** 

General discussion about documentation is probably another topic for the telecon (and over here: #210). I've just been informed that there is a recording of the latest DAWG meeting, so we can revisit that discussion in its entirety.

mts299, 2019-12-18:
Just a test reference when we start to investigate this issue more closely I found 20180903 unw fitacf data to be all partial records. I found this while trying to generate a summary plot with pyDARN and it failing to find `p_l`.

ecbland, 2020-03-14:
Closing this issue now that #280 has been merged 😄 

The last two items listed in the description of this issue are a bit out-of-scope (providing detailed definitions of the `qflg` and `x_qflg` parameters). To address this I've created a note in the "Documentation" project (https://github.com/SuperDARN/rst/projects/5#card-34590492)

closed 2020-03-14 by ecbland