SuperDARN/rst #231: Possible bug in rst idl library: superdarn/main.1.25/rprm.pro

agrocott, 2019-06-17:
I just tried to load in a random fitacf file: 20020202.0201.00.han.fitacf and the code crashed:

IDL: Conflicting data structures: structure tag,<INT       Array[24, 2]>.
IDL: Execution halted at: RADARDECODERADARPRM  275 /Applications/superdarn/rst-develop/idl/lib/superdarn/main.1.25/rprm.pro

Looking at the line in question:

  if (prm.mplgs gt 0) && (arrid[1] ne -1) then $
     prm.lag[0:prm.mplgs,*]=(*(arrvec[arrid[1]].ptr))[*,*]

it seemed as though the data range being specified is too long, i.e. it ought to be:

 if (prm.mplgs gt 0) && (arrid[1] ne -1) then $
     prm.lag[0:prm.mplgs-1,*]=(*(arrvec[arrid[1]].ptr))[*,*]

This fixed the issue for this file, however, testing this fix on another file (20120202.0201.00.sas.fitacf) caused the opposite problem, of the data range being too short. So, apparently not a fix, but also a problem with consistency!

I note this is related to pull request #121 - I am running 4.2 and seem to have the new version of the code detailed in this request.

Thanks,
Ade

egthomas, 2019-06-17:
@agrocott thanks for the message.  It seems like while the native IDL code does crash on that file, the C version of the code used by the DLMs appears to handle things correctly.

Examining that particular file more closely with `dmapdump`, the number of lags listed in the parameter block (`mplgs`) does not match the dimensions of the lag table array (`ltab`):

```
	short	"mppul" = 7
	short	"mplgs" = 18
        ...
arrays:
	short	"ptab" [7]
	short	"ltab" [19][2]
```

egthomas, 2019-06-17:
Actually, the error I encountered was because I tried calling the DLM and IDL versions consecutively and forgot to free the `fit` structure in between.  Just using the native IDL code to read the `fitacf` file we created for that day, eg

```
fitfp = FitOpenIDL('20020202.han.fitacf', /read)
while FitReadIDL(fitfp, prm, fit) ne -1 do begin
;    print, prm.time.hr, prm.time.mt, prm.time.sc
endwhile
s = FitCloseIDL(fitfp)

end
```

seems to work fine (where I've renamed each function in `fit.pro` to ensure the native IDL version is being called).  Likewise, reading the original dat file also seems to work for me:

```
rawfp = OldRawOpenIDL('2002020202h.dat')
while OldRawReadIDL(rawfp, prm, raw) ne -1 do begin
;    print, prm.bmnum, prm.time.hr, prm.time.mt, prm.time.sc
endwhile
s = OldRawCloseIDL(rawfp)

end
```

@agrocott could you share the `fitacf` file you're working with so I can try reproducing this issue?

agrocott, 2019-06-17:
@egthomas file attached.
Thanks.

[20020202.0202.00.han.fitacf.gz](https://github.com/SuperDARN/rst/files/3297188/20020202.0202.00.han.fitacf.gz)

egthomas, 2019-06-17:
@agrocott thanks - I took a quick look and the first thing I noticed is that this file appears to have been created using an older version of FITACF.  For example, comparing the values in the parameter block for the first record to a file I just generated from `2002020202f.dat` at Dartmouth:

| Evan's file | Adrian's file |
| --------------- | ------------------- |
| scalars: | scalars:
| 	char	"radar.revision.major" = 1 | 	char	"radar.revision.major" = 1
| 	char	"radar.revision.minor" = 66 | 	char	"radar.revision.minor" = 66
| 	char	"origin.code" = 1
| 	string	"origin.time" = "Mon Jun 17 17:10:14 2019"
| 	string	"origin.command" = "fittofitacf 2002020202f.fit"
| 	short	"cp" = 152 | 	short	"cp" = 152
| 	short	"stid" = 10 | 	short	"stid" = 10
| 	short	"time.yr" = 2002 | 	short	"time.yr" = 2002
| 	short	"time.mo" = 2 | 	short	"time.mo" = 2
| 	short	"time.dy" = 2 | 	short	"time.dy" = 2
| 	short	"time.hr" = 2 | 	short	"time.hr" = 2
| 	short	"time.mt" = 2 | 	short	"time.mt" = 2
| 	short	"time.sc" = 0 | 	short	"time.sc" = 0
| 	int	"time.us" = 0 | 	short	"time.us" = 0
| 	short	"txpow" = 9000 | 	short	"txpow" = 9000
| 	short	"nave" = 76 | 	short	"nave" = 76
| 	short	"atten" = 2 | 	short	"atten" = 2
| 	short	"lagfr" = 1200 | 	short	"lagfr" = 1200
| 	short	"smsep" = 300 | 	short	"smsep" = 300
| 	short	"ercod" = 0 | 	short	"ercod" = 0
| 	short	"stat.agc" = 0 | 	short	"stat.agc" = 0
| 	short	"stat.lopwr" = 0 | 	short	"stat.lopwr" = 0
| 	float	"noise.search" = 1.000000 | 	float	"noise.search" = 1.000000
| 	float	"noise.mean" = 57647.000000 | 	float	"noise.mean" = 57647.000000
| 	short	"channel" = 0 | 	short	"channel" = 0
| 	short	"bmnum" = 15 | 	short	"bmnum" = 15
| 	float	"bmazm" = 12.300000
| 	short	"scan" = -1 | 	short	"scan" = -1
| 	short	"offset" = 0 | 	short	"offset" = 400
| 	short	"rxrise" = 0 | 	short	"rxrise" = 0
| 	short	"intt.sc" = 7 | 	short	"intt.sc" = 7
| 	int	"intt.us" = 0 | 	short	"intt.us" = 0
| 	short	"txpl" = 300 | 	short	"txpl" = 300
| 	short	"mpinc" = 2400 | 	short	"mpinc" = 2400
| 	short	"mppul" = 7 | 	short	"mppul" = 7
| 	short	"mplgs" = 18 | 	short	"mplgs" = 18
| 	short	"mplgexs" = 0 
| 	short	"ifmode" = 0
| 	short	"nrang" = 70 | 	short	"nrang" = 70
| 	short	"frang" = 180 | 	short	"frang" = 180
| 	short	"rsep" = 45 | 	short	"rsep" = 45
| 	short	"xcf" = 1 | 	short	"xcf" = 1
| 	short	"tfreq" = 9955 | 	short	"tfreq" = 9955
| 	int	"mxpwr" = 4400000 | 	int	"mxpwr" = 4400000
| 	int	"lvmax" = 20000 | 	int	"lvmax" = 20000
| 	string	"combf" = ""
| 	int	"fitacf.revision.major" = 2 | 	int	"fitacf.revision.major" = 6
| 	int	"fitacf.revision.minor" = 5 | 	int	"fitacf.revision.minor" = 0
| | 	string	"combf" = ""
| 	float	"noise.sky" = 1.000000 | 	float	"noise.sky" = 11.000000
| 	float	"noise.lag0" = 0.000000 | 	float	"noise.lag0" = 13.000000
| 	float	"noise.vel" = 0.000000 | 	float	"noise.vel" = 0.000000
| arrays: | arrays:
| 	short	"ptab" [7] | 	short	"ptab" [7]
| 	short	"ltab" [19][2] | 	short	"ltab" [18][2]
| 	float	"pwr0" [70] | 	float	"pwr0" [70]
| 	short	"slist" [19] | 	short	"slist" [31]

While there are several differences, it looks like the most important one is that the lag table size in your file is different compared to the file generated using the current `develop` branch of the RST.  Converting our `2002020202f.dat` file to `rawacf` format and using `dmapdump`, the `ltab` array also has slightly larger dimensions than in your file:

```
arrays:
	short	"ptab" [7]
	short	"ltab" [19][2]
	short	"slist" [70]
```

So I'm not sure why your `fitacf` file ended up with a slightly smaller `ltab` array than is found in the `dat` file available from the BAS mirror.  Maybe it can be traced back to a difference in the software when writing on-site `fit` vs `dat` files?

egthomas, 2019-06-17:
I also just noticed the `offset` value in my `fitacf` file was changed from 400 in the original `dat` file to 0.  That doesn't seem like it should happen...

agrocott, 2019-06-19:
@egthomas 

Thanks for looking into this. Perhaps I’ll do some testing and see if this is a common issue or a one off... If it’s a common issue, then I guess my question is, do we treat this as a bug in an older version of FITACF, meaning this is a bug in the fitacf file, or is it just a difference between versions, in which case should rst be backward compatible with older versions of fitacf files?

ecbland, 2019-11-23:
@agrocott Do you have any updates regarding this issue? Was it just the `20020202.0202.00.han.fitacf` file, or have you been able to reproduce the issue with other FITACF files?

ecbland, 2020-02-13:
Closing this issue due to inactivity. @agrocott Please re-open if there is anything further to discuss.

closed 2020-02-13 by ecbland