SuperDARN/rst #233: Comparisons between pointers and integers introduced by #165

egthomas, 2019-06-20:
Pull request #165 introduced 11 instances of comparisons between a pointer and integer with code similar to 

```
} else if (rawfp->rawread==-2) {
```

These can be found in the following binaries:

- dattorawacf
- test_raw
- cat_raw
- make_fit
- make_fitex
- make_fitex2
- make_info
- make_lmfit
- raw_cp
- rawstat
- trim_raw

These were noted in [Dec 2018](https://github.com/SuperDARN/rst/pull/165#issuecomment-444288336) and [Mar 2019](https://github.com/SuperDARN/rst/pull/165#issuecomment-475436174).  It would be nice if these were addressed, particularly since the compiler warning for `trim_raw` is the last thing someone sees when compiling the RST without IDL.

egthomas, 2019-06-28:
So since it has been removed from the project, is this not going to be addressed before the 4.3 release? It doesn't seem ideal to distribute code with known problems that we as developers introduced.

ecbland, 2019-06-28:
This looks like something that can be fixed fairly easily in time for the v4.3 release, and I agree that we should avoid distributing code with known problems. Do you have time to implement the required changes @egthomas ?

ksterne, 2019-06-28:
@egthomas, @ecbland, I initially removed this from the project since there wasn't an open pull request here.  I think until I see a pull request on it, this could potentially slow the project down.  I understand the hesitation here since the warning is the last thing that could be seen while compiling.  However, the warnings do not cause compilation to fail and do not seem to have an effect on data quality.  So, I evaluated this as something that could take a while to resolve and doesn't seem to be critical.

As we noted in #165, there's a bigger issue here of exit/return codes for error handling that doesn't seem to be well handled.  From what I could see, the solution to this issue (233) will need some significant rewriting of the code in order to properly set the rawread variable and return an error code that handles things properly.  Diving into this isn't something I can handle here while trying to get months old code into a new release.  Maybe there's a different solution I'm not seeing that could be easier though.

So, I think there's two solutions here:
1. We wait for a pull request to be generated from this issue that is tested and accepted/merged before starting the 4.3 release process.
2. We continue on with the 4.3 release while this issue is still being worked on and once it is resolved, we can have a 4.3.1 release that can include this.

Sorry, I know some of this is out of scope for the issue and should be more so in #237, but there's some detail on potential solutions as well.  

Let me know thoughts on which would be preferred; I'll go with the group even though I recommend the later solution.

ecbland, 2019-06-29:
@ksterne Ok, if this issue is complicated to fix then it will indeed delay the project. I'll let @egthomas comment further since he opened the issue.

egthomas, 2019-07-01:
> However, the warnings do not cause compilation to fail and do not seem to have an effect on data quality.

I respectfully disagree with this sentiment - the compiler is providing this warning for a reason, and we as developers should be reducing the number of errors and warnings in the code, not introducing more.  And the code compiling does not guarantee expected behavior on all systems.

Ideally such issues should be identified and addressed before the pull request is approved, but since that isn't always realistic, I think at a minimum they should be fixed before an official release is distributed to the community.

ksterne, 2019-07-01:
I'll admit that I caused these problems, so I'll offer another solution here.  Basically revert #165 until we can better document the code and how values/variables are being passed between functions/files as well as return and exit codes.  I've reverted that pull request in the `bugfix/revert_warning_message_bad_num_bytes` branch.  I need to check that it's cleared the warnings later tonight or later this week.  @egthomas, does this seem like a viable solution?

asreimer, 2019-07-01:
Isn't the simplest solution to simply dereference the pointers in the comparisons instead of reverting? Basically amounting to adding 11 `*`, one in each of the offending locations?

egthomas, 2019-07-02:
`rawfp->rawread` is a function pointer and not a pointer, so I do not believe this implementation of a `-2` error code was ever truly functioning as intended.

@ksterne on that branch, many more changes have been reverted beyond just the `num_bytes` warning messages.  Specifically, all of the command line option error handling for those binaries has also been removed.

mts299, 2019-07-02:
Would it be possible to let me take a look through my software developers eyes if I can be of assistance? 

Sorry for the delay, long weekend in Canada and as said before the time zone change makes it difficult to respond quickly. 

I would hate to see code reverted purely because of a warning message. It may be the path of least resistance but doesn't mean it is the most productive path.

asreimer, 2019-07-02:
The problem is that sometimes `rawfp->rawread` points to a function and other times, depending on `num_bytes`, it is set to an integer. See here: https://github.com/SuperDARN/rst/blob/develop/codebase/superdarn/src.lib/tk/oldraw.1.16/src/rawopen.c#L95

I think to fix this, the OldRawFp structure needs to have another field added to it to handle error codes. Anyone else have ideas for solutions?

mts299, 2019-07-02:
Just to cycle back where someone mentioned we need to address warnings before pushing or releasing, RST currently compiles with 127 warning messages. 

Anyways, as @asreimer has mentioned rawread is a function pointer https://github.com/SuperDARN/rst/blob/444abd4f38ff18bb6f9a25b6272254fe8af8b4d0/codebase/superdarn/src.lib/tk/oldraw.1.16/include/oldrawread.h#L50

Partly my fault for not seeing it in my code review of #165. 

So promoting error codes :p I made a fix to this issue by just adding extra value in the struct for error. Granted this is a quick fix to solving the warnings, we should discuss a more elegant method that allows for better maintenance.

asreimer, 2019-07-02:
Just to clarify, @egthomas mentioned rawread is a function pointer and I clarified that it is only sometimes a function pointer as of #165. :)

Otherwise, sounds good! I think the extra value added to the struct is moving in the right direction for supporting proper error code handling. @ecbland @ksterne what do you two think of that?

egthomas, 2019-07-02:
@mts299 thank you for implementing a solution to this issue.

> Just to cycle back where someone mentioned we need to address warnings before pushing or releasing, RST currently compiles with 127 warning messages.

To further clarify - no one said we can't release the RST without warning messages; what I said was we should not be introducing new ones:

> ... we as developers should be reducing the number of errors and warnings in the code, not introducing more.

which is a stance that seems to fit the "low conflict, high reward" criteria.  Over the past several years we've spent considerable time and effort reducing the total number of compiler errors and warnings from the nearly 500 we started with (eg, #108, #137) down to the current number of ~130.

mts299, 2019-07-02:
Thanks @asreimer and @egthomas for pointing out the importance of error codes and the importance of addressing warning messages. 

>Just to clarify, @egthomas mentioned rawread is a function pointer and I clarified that it is only sometimes a function pointer as of #165. :)
Sorry, in my flurry of getting up this morning and seeing all the emails I just saw @asreimer name and function pointer. Good catch @egthomas. 

Just to keep things on track, please test out my PR #241 so we can close this issue :) 

And we can move the discussion of error handling to issue #240 that @asreimer created :+1:

asreimer, 2019-07-02:
I tested #241. It works, so I'm closing this.

closed 2019-07-02 by asreimer

ksterne, 2019-07-03:
A mountain of thanks to @asreimer and @mts299 as they are better at coding than I am.  I am still a bit of a c n00b.

asreimer, 2019-07-03:
Teamwork! 💪