SuperDARN/rst #236: Skynoise rounding error

ecbland, 2019-06-24:
While testing #234 I noticed that the skynoise parameter is not identical when I generate a (new) fitacf file from an (old) dat file in two different ways: 
```
make_fit -old 2002020202f.dat 2002020202f.fit
fittofitacf 2002020202f.fit > 2002020202f.fitacf
dmapdump 2002020202f.fitacf | grep noise.sky
```
does not produce the same as 
```
dattorawacf 2002020202f.dat > 2002020202f.rawacf
make_fit 2002020202f.rawacf > 2002020202f.fitacf
dmapdump 2002020202f.fitacf | grep noise.sky
```
In the first example, the skynoise parameter is always rounded down to the nearest integer. The other fitted parameters are also slightly different, but I'm assuming for now that the skynoise is the cause. 

It would surprise me if nobody has noticed this before. Can anyone enlighten me before I try to fix it?

egthomas, 2019-06-24:
I think these are the offending lines of code:

```
  /*  Now transfer the noise parameters */

  r.r1.r_noise_lev = (long) fptr->noise.skynoise;
  r.r1.r_noise_lag0 = fptr->noise.lag0;
  r.r1.r_noise_vel = OldFitCMP16(10,fptr->noise.vel);
```

https://github.com/SuperDARN/rst/blob/develop/codebase/superdarn/src.lib/tk/oldfit.1.25/src/fitwrite.c#L342-L346

The old `fit` format file played a lot of compression games I think due to the fixed size of the header block.  I'm not sure if this is something we can / should modify.

pasha-ponomarenko, 2019-06-24:
I tend to agree with Evan.

Dr Pavlo (Pasha) Ponomarenko
Senior Analyst, Research Computing
Academic and Research Technologies
University of Saskatchewan

Phone   1(306) 966 4003
Email     pasha.ponomarenko@usask.ca

https://github.com/SuperDARN/rst/blob/develop/codebase/superdarn/src.lib/tk/oldfit.1.25/src/fitwrite.c#L342-L346

[https://avatars3.githubusercontent.com/u/23531490?s=400&v=4]<https://github.com/SuperDARN/rst/blob/develop/codebase/superdarn/src.lib/tk/oldfit.1.25/src/fitwrite.c#L342-L346>
SuperDARN/rst<https://github.com/SuperDARN/rst/blob/develop/codebase/superdarn/src.lib/tk/oldfit.1.25/src/fitwrite.c#L342-L346>
github.com
Radar Software Toolkit (RST). Contribute to SuperDARN/rst development by creating an account on GitHub.

ecbland, 2019-06-24:
Ok, thanks. This suggests that it's better to convert from `dat->rawacf` before calling `make_fit`, instead of using `make_fit -old` directly. This would be good to add to the documentation.

egthomas, 2019-06-24:
@ecbland generally yes, I agree users should go from `dat` -> `rawacf` -> `fitacf` as you indicated.  I think we've fixed a lot of the bugs associated with that procedure over the last couple of years (eg #139, #146, etc).

Apparently though, due to the lossy compression applied to the `dat` files produced on-site at the radars, a `fit` file produced on-site cannot be perfectly reproduced from the associated `dat` file.  This has become less of a consideration because we no longer distribute the on-site `fit` files via the data mirrors (which furthermore were produced using much older and typically unknown versions of FITACF).

closed 2019-06-24 by ecbland