SuperDARN/rst #237: Next release?

egthomas, 2019-06-24:
So @ecbland and @ksterne may have already worked this out amongst themselves, but I was just curious about the plan for the next release.  My personal feeling is that enough new features have been merged since `v4.2` (eg, the Shepherd [2017] elevation algorithm, extended `time_plot` and `field_plot` capabilities, the new `elevation` and `cnvmodel` libraries, etc) to warrant an eventual new minor release as `v4.3` (as opposed to just a patch release).

aburrell, 2019-06-24:
That is also my impression, but I defer to the WG heads :)

ecbland, 2019-06-24:
@egthomas you read my mind...thanks for bringing it up! I agree that we are now in a position to skip the patch release and instead go for a minor release. @ksterne will be overseeing the releases so I'll let him comment further.

ksterne, 2019-06-25:
@egthomas, this is likely true.  There's been a recent flurry of updates here that likely have pushed this into a new minor release.  I'd started up the patch hoping/thinking we could get a few simple PRs finished out.  Sorry I've been slow on this as being slowed down by 2.5 weeks or vacation and lots of travel.  I'll try to get this going in the next few weeks.

ecbland, 2019-06-25:
Looking through the "to do" list on the v4.3 project page, there are some long-standing issues that are unlikely to be resolved in reasonable time for a minor release. This is not a problem, but we should remove these from the v4.3 project to get a better view of the current status. I'm looking in particular at issues #3, #32, #135 and #191.

@egthomas Is it time to revisit #188?

Note also that #210 and #215 are more general discussions about documentation, an ongoing project in itself. I'm tempted to remove these from the v4.3 project due to their large scope. @mts299 - Is there a specific goal for the documentation which is achievable within a few weeks (with help)?

ksterne, 2019-06-26:
I should note that I want to get a release out the door in the next month (meaning a release is actually made) as we are woefully behind on making a new release.  So, I may end up descoping some of the v4.3 project in order to achieve this.  Things that are left out and be made into a patch release or another minor release if needed.

mts299, 2019-07-02:
Sorry for the delay @ecbland I think #215 is achievable since I have been getting help from @asreimer and @billetd. Once we get the installation guide finished I think we can make a PR on that alone with the RST HowTo's

The DMAP format will need some TLC, which I may have some time this week to tackle it. 

One thing I may suggest is that documentation doesn't necessarily need to be pushed into develop but could be pushed to master if the content is not addressing to specific code changes. Like #141 could have been pushed to master to help current users of the time with the current RST instead of waiting for a release. 

This may help the slow progression of switching to a new documentation system.

ecbland, 2019-07-03:
Thanks @mts299, that's a good plan. And I agree that many of these documentation updates can go straight to `master` in the future. 

@ksterne Perhaps it's a good time to set a deadline for making the v4.3 release?

ksterne, 2019-07-08:
@ecbland, (and all) sorry I've maybe not taken the lead on things too much here; I'm finally back in the office (longer than I have in the last week than in the last 2 months it feels).  We've got 2 more things to tackle here and then I'll get out a deadline for these...or maybe the other way around.  I'm hoping to help out with #186, but lots to do and catch up on here. But yes, a 4.3-release branch should be coming soon.  I'd kindly ask everyone to try as much as they can to focus on the two "in progress" cards on the 4.3 project.

ksterne, 2019-07-31:
Hey all, since it seems as though #246 is close to complete, we'll see if we can get it in.  I don't think I'll let it hold up a release branch though.  So, still waiting on #247, which seems very close, and then we'll be ready for a release.  To put things into context, I'd hoped to have had a patch release about a month and half ago and we're about 3 weeks away from coming up on a year since our last minor release.  I know lots of us have lots going on and I'm certainly appreciative of the work we're doing.

mts299, 2019-08-01:
PRs #246 and #247 are merged! A release now can be made :+1: 

Side note: if you could add a LICENSE.md in the release that would be good as well since we seem to refer to a lincense.txt in our code but not sure we actually have one. I don't think that addition needs a PR.

ecbland, 2019-08-01:
@ksterne Are all the hardware files up-to-date on the `develop` branch? They are not identical to the `hdw.dat` repo, and I'm not sure which repo is ahead of the other (perhaps it's a mix of both?)

ksterne, 2019-08-01:
Great work on getting those last 2 merged in!  I'm on travel today and need to catch up on somethings tomorrow, but if I don't get to it tomorrow I'll get to looking at the `hdw.dat` repo and create a release branch first thing Monday morning.

ksterne, 2019-08-05:
Sorry it took me a bit here.  You are right that there are "differences" between the two sets of files on the repos.  I did a checksum of the file contents and here are the ones that are mismatched and the summary of the differences:

- `hdw.dat.bks`: random whitespace difference
- `hdw.dat.cly`: rst repo file has longer explanation header whereas hdw.dat repo has much shorter header.  Contents (non-commented lines) are the same though.
- `hdw.dat.ekb`: rst repo file has redundant radar and differing whitespace
- `hdw.dat.fir`: rst repo file differs only in whitespace and an additional comment line
- `hdw.dat.hkw`: rst repo file differs only in whitespace
- `hdw.dat.kap`: seems to differ on the whole file, but the main contents are the same
- `hdw.dat.kod`: rst repo file differs only in whitespace
- `hdw.dat.mcm`: **rst and hdw.dat repos files differ on tdiff value**
- `hdw.dat.pgr`: seems to differ on the whole file, but the main contents are the same
- `hdw.dat.rkn`: same as `pgr`
- `hdw.dat.sas`: same as `pgr`
- `hdw.dat.sps`: rst repo file differs only in whitespace
- `hdw.dat.sto`: same as `sto`
- `hdw.dat.zho`: rst repo file differs only in one whitespace

So, we're mostly OK here as the contents are all good.  I think where the files differ on the whole file is from a different file format or editor being used for each of the files.  The one that does need checked on is the `hdw.dat.mcm` file.  I'll work with @wabristow to figure out which value of tdiff is right.  For reference the commit to this repo is from Oct. 2018 and the hdw.dat repo commit is from Feb. 2019.  So, it's possible we need to update the rst repo file.

Unless there's strong feelings here, I'll make a release branch tomorrow and will update the hdw.dat file in the release branch while that branch is going under the "testing period".  If anyone does have strong feelings but can't quite get to looking into this yet, then at least let me know before tomorrow you'd like to take a second look.

ksterne, 2019-09-06:
I'm always a fan of the person that opened the issue to close it out.  Since the version 4.3 is now through, @egthomas, do you want to close this issue?