SuperDARN/rst #240: Using different exit codes for different errors

asreimer, 2019-06-25:
With #239, new a new exit code has been introduced to signal when there is a error reading a file.

I'm sure @mts299 and others who know what they are doing have better resources/info to share, but here's a guide to exit codes for bash scripts: http://tldp.org/LDP/abs/html/exitcodes.html, but there isn't really a guide for C that I know of (except to use 0 and 1). It might be helpful to think about how exit codes differ on Windows and Linux: https://en.wikipedia.org/wiki/Exit_status#POSIX

asreimer, 2019-06-25:
Running a `grep -lr "exit(0)" *` and `grep -lr "exit(-1)" *` on the codebase, returns results, but `grep -lr "exit(-2)" *` and any other negative exit number doesn't seem to.

What's even better, is that `grep -lr "exit(1)" *` and `grep -lr "exit(2)" *` also return results, meaning we already have a mix of exit codes in the codebase. For example, exit code 2 is used for "unknown host" in `codebase/general/src.bin/tcpip/buffer_server.1.14/buffer_server.c` and exit code 1 is used for socket errors. Yet in `codebase/analysis/src.bin/aacgm/aacgmeval.1.2/aacgm.c`, exit code 1 is used for a file reading problem.

So we need some consistency probably.

mts299, 2019-06-25:
@asreimer you are correct! In C there is also a library typically used for storing error values in that other functions like `fopen` uses when it has to return a non-int. [errno.h](https://www.tutorialspoint.com/cprogramming/c_error_handling.htm) is the library you import and have access to the errno macro which stores the error code. Then you can check that value after every function call instead of the return value for the error. 

This way you can return a more meaningful error code and not try and let the developer figure out what the NULL pointer was supposed to mean. You can even use predefined macros

asreimer, 2019-06-25:
Ya, we should definitely be using errno.h at the very least then!

mts299, 2019-06-25:
* sorry for the close and reopen, fingers slipped :p and never got to finish writing my message.

mts299, 2019-06-25:
@asreimer here are some codes related to [Windows](https://docs.microsoft.com/en-us/windows/desktop/debug/system-error-codes--0-499-).

mts299, 2019-06-25:
Ends up we are already using other values than -1 for errors and I think this was brought up in #165  when @ksterne had to identify the difference between a "file does not exist" vs. "zero length array". 

https://github.com/SuperDARN/rst/blob/444abd4f38ff18bb6f9a25b6272254fe8af8b4d0/codebase/superdarn/src.lib/tk/radar.1.22/src/rprm.c#L185-L186

mts299, 2020-07-16:
So I might add this to the developer guidelines and then maybe we should make this a project on how to attack it so we can do it in smaller steps? @ecbland ?

ecbland, 2020-07-16:
@mts299 Yes that sounds good. I think this conversation went stale so it would make sense to address it in the developer guide

mts299, 2020-07-16:
Yea I feel it went stale because there were no set error code decisions like:
- should we use errno.h (which I say yes)?
- if we define our own error codes what are they?

So at least having this set in a guideline will pave for the future devs to know what to do.

mts299, 2020-07-16:
I am going to close this, open a new issue of "defining error codes" to revitalize the discussion, and put it on the TODO for guidelines.

closed 2020-07-16 by mts299