SuperDARN/rst #243: Unused `dyno` variable in igrf_v2/igrflib.c

asreimer, 2019-07-02:
It appears that the lines:

    int dyno;

and 

    dyno = dayno(igrf_date.year,0,0,&(igrf_date.daysinyear));

in `codebase/analysis/src.lib/igrf_v2/igrf.1.0/src/igrflib.c` are unused.

Safe to remove? Relevant to #242 because `dyno` is throwing unused variable compiler warnings.

asreimer, 2019-07-02:
It's clearly unused in the function, but is that intended or a bug?

ecbland, 2020-02-11:
@asreimer Is it ok to close this issue since we agreed not to modify the IGRF/AACGM-v2 code in #242?

asreimer, 2020-02-11:
If the DAWG has decided that external libraries are not to be modified then this issue should be closed.

There should be a discussion about whether "external" libraries should even be included in RST to begin with in regards to how they impact the maintainability and consistency of the codebase. There are becoming an ever increasing number of edge-cases that complicate the development process. For example, why is it ok to squash compiler warnings and modify FITACF3 when adding it to RST, but not AACGM? Just some things to think about.

closed 2020-02-11 by asreimer

ecbland, 2020-02-11:
Thanks @asreimer. We received a specific request to not modify the AACGM-v2 code, but I agree that we need to have a broader discussion about the external libraries. This probably can't be solved productively on github so I'll put it on the agenda for our next meeting.