SuperDARN/rst #244: `err` set but unused in mlt.c

asreimer, 2019-07-02:
In `astmlt1`, a variable `err` is defined and then set, but it is never checked. It is set by a call to `convert_geo_coord`, which returns 0 if successful and a bunch of different numbers if it isn't.

Question is:
1) shouldn't we be checking this?
2) if not, we can just remove `err` everywhere it appears, and not bother to catch the return value of `convert_geo_coord`, or
3) if we don't want to check it, but want to catch the return value, then we could tell the compiler to ignore this case with a pragma: http://gcc.gnu.org/onlinedocs/gcc/Diagnostic-Pragmas.html

For 3, something like:

    #pragma GCC diagnostic ignored "-Wunused-but-set-variable"
    int err;

works great.

So, what do?

asreimer, 2019-07-02:
Huh, same thing with `codebase/imagery/src.lib/szamap.1.11/src/szamap.c` with the variable `s` defined on line 56.

This seems to be a common problem.

mts299, 2020-09-17:
addressed in PR #337

closed 2020-11-02 by mts299