SuperDARN/rst #251: map_addmodel states Grid file not found

mts299, 2019-08-15:
Forgetting where I was and just hitting the up arrow key I used `map_addmodel` in the wrong directory and got this error message:
```bash
Grid file not found.
```
At first, I was really confused and thought the command was looking for grid files but realized it is a map file generating binary. 

So I am wondering if we should change it to: 
```bash
Map file not found
```
Or just to keep it simple and consistent with other binaries like `make_fit`:
```bash
File not found
```

Just to be more intuitive to the non-superDARN user.

aburrell, 2019-08-15:
I vote for "File not found".

ksterne, 2020-01-07:
@mts299, care to close this now that #258 is merged?

closed 2020-01-07 by mts299