SuperDARN/rst #257: FITACF 3.0 still not using Simon's elevation algorithm?

mts299, 2019-11-13:
Where are we at for getting Simon's elevation algorithm as a default or at least an option for Fitacf 3.0 to use? 

I remember I wrote a code solution for it but I guess it never got pushed and not we have two fitacf algorithms using two different elevation algorithms by default :/ 

@egthomas thoughts?

ecbland, 2019-11-15:
Thanks @mts299 for putting this back on the agenda! We missed this in the telecon and it should be a priority for the next release. I'm happy to take a look at this over the next few days (starting with your code from #150), or would you prefer to do it yourself?

mts299, 2019-11-18:
@ecbland I have no problem cleaning up the code again to get it to work with RST 4.3 if things have shifted. However, one thing we may need to address before attacking that code is the inconsistencies in constant values #238; this is when I first noticed it when Fitacf 3.0 used a different version of the speed of light than fitacf 2.5. With my solution in #150, I think I had to use the fitacf 2.5 speed of light for both Fitacf versions. However, when testing with the older version of RST, this made the change in Fitacf 3.0 stick out. 

I imagine the following needs to be done: 
- [ ] implement a constant standards library (merge)
- [ ] re-open PR #150 (or make a new one) based on `develop` changes from the above point
- [ ] Test the above PR against `develop` (since constants should be standardized)

ecbland, 2019-11-18:
@mts299 Ok, let's focus on #238 first.

mts299, 2020-06-01:
Closing this as things are changing. Will create a new Issue when plans are made.

closed 2020-06-01 by mts299