SuperDARN/rst #260: How to read the fitacf file directly ?

FancystarWang, 2019-12-19:
It's very easy to make some plots from the data by using RST. But when I want to read the data ,rather 
just plot the rti_plot or cnvmp , there is not a clear guide to tell the new one like me  how to make some detail research. Reformating fitacf to ascii is not a good choice, so must I change the C code to  get the detail data? Well, I hope it will be easier.

asreimer, 2019-12-19:
Hey @FancystarWang, there are essentially 2 options:
1) use the dmap library in C to read the fitacf file contents, see https://github.com/SuperDARN/rst/tree/master/codebase/general/src.lib/dmap.1.25
2) use python to read the fitacf file contents

If you want to use python instead, there are 2 options:
1) pydarn, which is under active development, but the file reading is working beautifully: https://github.com/SuperDARN/pydarn
2) using backscatter: https://backscatter.readthedocs.io/en/latest/backscatter.dmap.html#reading-and-decoding-data

I would start with pydarn and try the usage example given here: https://github.com/SuperDARN/pydarn/blob/develop/README.md

mts299, 2019-12-19:
Here is the most updated documentation on pyDARN: https://pydarn.readthedocs.io/en/doc-user/
Including up to date installation guidelines and tutorials on reading SuperDARN files.

FancystarWang, 2019-12-23:
When I read codebase，I found  a C file named fit to ascii . It is not a finished file but useful .  And I've tried the pyDARN,it's useful too.Thank you~

closed 2019-12-23 by FancystarWang