SuperDARN/rst #262: map_plot option "-def" not listed as an option.

ksterne, 2020-01-08:
While starting to look into #207, I needed to learn how to use `map_plot`.  Turned to the documentation and quickly noticed that the `-def` option is described and listed in the Description section, but not listed in the Options section.  It's awesome there's a larger description of this option that makes things friendly for first-timers, but I would think it should be in the Options section too.  Am I missing something?

Not a top priority and might be complicated if we're transitioning how our functions are documented, but thought I'd capture this observation.

ecbland, 2020-01-08:
I assume you're talking about this documentation: https://superdarn.github.io/rst/superdarn/src.bin/tk/plot/map_plot/index.html

The `-def`option is also missing from the `hlpstr.h` file (displayed when you type `map_plot --help`), and also from the readthedocs documentation. 

This option is also available in `grid_plot`, where it is correctly documented in the Options section: https://superdarn.github.io/rst/superdarn/src.bin/tk/plot/grid_plot/index.html

so it probably just got lost from `map_plot` along the way.

egthomas, 2020-01-08:
https://github.com/SuperDARN/rst/blob/ed55487a6315c7353687ba2af327cf3ab15d919c/codebase/superdarn/src.bin/tk/plot/map_plot.1.12/map_plot.c#L692

https://github.com/SuperDARN/rst/blob/ed55487a6315c7353687ba2af327cf3ab15d919c/codebase/superdarn/src.bin/tk/plot/map_plot.1.12/map_plot.c#L762-L776

ksterne, 2020-02-18:
This is solved by #264 so going to close this down for now.

closed 2020-02-18 by ksterne