SuperDARN/rst #263: Jiamusi Radar

mts299, 2020-01-21:
Maybe one more thing to consider for the next release is adding Jiamusi Radar to the radar.dat file so we can process the data. Currently, I cannot process it to FITACF because of the commission issue that was brought up before #239. 

However, this data has been cleared and now part of the data distribution.

ksterne, 2020-01-21:
It's still not been clear to me when the commission date is and/or the acceptance by the PIs.  I think there was a discussion about this at the AGU meeting, but I haven't followed up with my PI on it.  I'll try to do that here soon.

mts299, 2020-01-21:
Okay, if not can we look at making a warning message and not stop the ability to process the data?

ecbland, 2020-02-11:
@ksterne Any update on this? I noticed that the JME hardware file is already in the `hdw.dat` repo, so we should copy that over to RST. We still need to find out the start date to use in the `radar.dat` file.

ksterne, 2020-02-20:
I got word back and got this info in #276.  So this is another one that I think I'd leave to @mts299 to close, but I will for now.  Marina, feel free to open it back up or the other one if needed.

closed 2020-02-20 by ksterne