SuperDARN/rst #268: What's the radius of the Earth?!

ecbland, 2020-02-11:
During the DAWG telecon on 21 January we discussed how mathematical and physical constants are handled within RST (#238). We agreed that mathematical constants (e.g. pi) would be defined using `math.h`, and physical constants (e.g. speed of light, earth radius) would be defined in a separate file in RST.

This issue is specifically about the value of the **Earth radius**. So far I've found two definitions of the Earth radius in RST:

1. Earth's mean spherical radius, RE=6371.2km. This value is used in the IGRF, so RST uses it for anything related to IGRF/AACGM. 

2. Earth's polar radius, RE=6356.779km, corresponding to the Earth radius at 88°N geodetic latitude. This value is used by the field of view mapping routines (`RPos*`/`RadarPos*`), which convert range/beam coordinates to geographic coordinates. 

I suspect that (2) is used in the `RPos` routines because the SuperDARN radar fields of view cover higher latitudes where the earth radius is slightly smaller. But this will always be an approximation and it might not really matter which value is used.

I've created this issue to highlight the fact that multiple definitions exist and to give people an opportunity to offer their expertise. @mts299 is planning to create a file of physical constants in RST so that they don't need to be defined separately in each routine (see #238), and has asked for advice on what values should be used. I suggest that we create two separate definitions of RE for now, and then take the opportunity during testing to see if there is any practical difference between the two values.

mts299, 2020-02-11:
@DannoPeters

aburrell, 2020-02-11:
> I suggest that we create two separate definitions of RE for now, and then take the opportunity during testing to see if there is any practical difference between the two values.

I agree with the above statement made by @ecbland

egthomas, 2020-02-11:
[rmath.h](https://github.com/SuperDARN/rst/blob/master/codebase/base/src.lib/math/rmath.1.8/include/rmath.h) has historically been the source of mathematical constants for the RST.  That seems like a logical place to consolidate the constants already defined there like C or pi for a minimal impact on the existing software.

I would advocate for some caution when changing the constants defined in externally developed libraries (AACGM, IGRF, AstAlg, etc).  For example, by changing the AstAlg constants you might be making them more consistent with those used by other RST libraries while also making the results inconsistent with those obtained by anyone else using that published library.

Regarding the Earth radius values, we did some related tests a few years ago.  The variation in geodetic latitude along a radar beam for Northern Hemisphere sites has a maximum value of ~10 km (BKS beam 23), minimum of ~0.7 km (BKS beam 0), and a mean and median of ~5 km.  That maximum variation in Earth radius from the nearest to furthest range gate of ~10 km is comparable to the difference between 6356.8 km (88 degree geodetic north) and 6371.2 km (mean spherical radius) cited above, which at the time we decided was negligible on the calculated range given the inherent assumptions in either the iterative RadarFldPnt routines or the simplified spherical Earth geometry used by Chisham et al [2008] and others when calculating ground ranges or virtual heights from measured slant ranges and elevation angles.  But again, it's been a while so others are welcome to double check the math on that.

ecbland, 2020-02-11:
> I would advocate for some caution when changing the constants defined in externally developed libraries (AACGM, IGRF, AstAlg, etc). 

Good point @egthomas. We have agreed not to modify the AACGM-v2 code anyway, but we should be careful with the other external libraries too.

mts299, 2020-02-11:
Sorry, this is out of scope, but why are we copying libraries' source code into the RST source code?
 
As long as AACGM, IGRF, AstAlg, etc. pull their constants from wherever they define them and do not reference `rmath.h` I don't think there will be an issue. 

> I suspect that (2) is used in the RPos routines because the SuperDARN radar fields of view cover higher latitudes where the earth radius is slightly smaller. But this will always be an approximation, and it might not matter which value is used.

I agree with this as well @ecbland, might be something @DannoPeters would be interested in testing or being involved.

asreimer, 2020-02-11:
I suggest:
1) defining the Earth's radius as the one used in the geocentric coordinates. This is what most people will expect when you talk about the radius of the Earth (so mean Earth radius), or
2) fundamentally the question posed here arises because the `RPos*` and `RadarPos*` functions make approximations that are really not necessary. If the concern is about precision, why not update those function so they do the math rigorously for a WGS84 datum geodetic coordinate system instead?

First, the DAWG could do 1 and work towards implementing 2.

pasha-ponomarenko, 2020-02-11:
This is a rare occasion I feel like adding my five Aussie cents: 
(1) the difference of (14.4 km/6371.2 km)=0.23% in R_E would cause the same order of magnitude difference in geolocation. 
(2) agree with Ashton that the average (mean?) user would expect us to use the average (mean?) R_E.

egthomas, 2020-02-11:
It looks like the various `RadarPos` functions do use WGS-84 for all internal calculations (I remember updating this from IAU-64 to WGS-84 a few years ago).  The only place the hard-coded `re` variable is used is for calculating a 0.5-hop virtual height assuming the input altitude is an elevation angle (because it is < 90):

https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.lib/tk/rpos.1.7/src/cnvtcoord.c#L364

However even that value appears to be overwritten later in `fldpnth` for certain ranges with the standard virtual height model (150 < r < 800 km) or for all ranges with the Chisham model.  Furthermore, the existing C software has no way of using elevation angles as input to the RadarPos functions (ie the virtual height is always set to a fixed value when gridding or plotting according to the virtual height model selected).  The only way users can access that functionality is by calling the DLM from IDL.

egthomas, 2020-02-11:
Some back of the envelope calculations for ground range assuming 0.5- or 1.5-hop scatter at ranges of 1000-2500 km and a virtual height of 300 km using the two different values for Earth radii (6371.2 or 6356.779 km) are on the order of 0.1 km or less.

ecbland, 2020-02-11:
Thanks for the comments everyone! This is the type of feedback I was hoping for when I created the issue 😄 

My suggestion to retain both definitions was to help separate the science question ("what value(s) of RE should be used?") from the implementation of better coding practices (define the constants in a central location, #238). But based on @egthomas's and @pasha-ponomarenko's comments it seems we are happy to move to a single definition of RE=6371.2km immediately. I think  @asreimer is right that most users probably think that we are already using this value for everything. 

@aburrell You agreed with my original suggestion to have two values, but do you have any further comments now that others have weighed in?

pasha-ponomarenko, 2020-02-11:
Just wanted you to avoid wasting time in order to prove that 0=0.
:-)

aburrell, 2020-02-12:
Given:
> Furthermore, the existing C software has no way of using elevation angles as input to the RadarPos functions (ie the virtual height is always set to a fixed value when gridding or plotting according to the virtual height model selected). 

and everything else, there doesn't seem to be much point in adding a negligible amount of precision by keeping the polar Terrestrial radius.  I am happy to move forward with a single definition of the mean earth radius.  I would suggest putting some comments next to it, saying why we're using this one so that future DAWG members understand the reasons behind the choice.

ecbland, 2020-02-13:
We seem to have reached a consensus to move to a single definition of the Earth radius throughout RST: 6371.2km. So I will close the issue - thanks everyone!

closed 2020-02-13 by ecbland