SuperDARN/rst #271: rst/.profile.bash command not working

snmhoque, 2020-02-12:
the command shows permission denied. The sudo rst/.profile.bash command gives error : command not found. please advise

ecbland, 2020-02-12:
@snmhoque Could you please provide some more information about what you are trying to do? Are you trying to install RST?

During the installation you should add the following lines to your `~/.bashrc` profile (open this file in a text editor):
```
# bash profile for rst
export RSTPATH="INSTALL LOCATION"/rst
. $RSTPATH/.profile.bash
```
(and replace "INSTALL LOCATION" with the full path to the rst directory. For example `/usr/local` or `/home/[username]`). You then need to run `source ~/.bashrc` from the command line to reload the environment variables. You should not need to use sudo. 

https://radar-software-toolkit-rst.readthedocs.io/en/latest/user_guide/linux_install/#installation

ecbland, 2020-02-18:
Closing this issue due to lack of response. 
@snmhoque Please reopen the issue if you still need help with this.

closed 2020-02-18 by ecbland