SuperDARN/rst #272: Rlease Guidelines and Authorship

mts299, 2020-02-13:
To keep the PR #269 focused on "how to contribute to RST as a new user," I have moved @egthomas discussion to here on the discussion/todo to make a release guideline and formalized way of determining authorship. Thus we are following our communication guidelines. 

> It may not need to go in the readme or be decided with this pull request, but the procedure for generating the Zenodo/DOI citation (ie author list and order) for each release should also be formalized. Are all past authors always included in the author list? Only those who contributed since the previous release? Some middle ground between those two options? That's why I haven't updated the Zenodo page for the most recent patch release - unlike previous minor releases the approach was not quite as clear.

ecbland, 2020-02-13:
Thanks @egthomas for raising this issue. I'll put it on the agenda for our next telecon, but comments are welcome in the meantime. We do have [release guidelines](https://homepage.usask.ca/~pbp672/RST_release_guidelines.pdf) but they do not address authorship.

ksterne, 2020-09-01:
I came across this tab being open, likely as it was something I knew we needed to work on.  I think this topic has since been divided into 2 topics in the DAWG repo: https://github.com/SuperDARN/dawg/pull/5 and https://github.com/SuperDARN/dawg/pull/6 . Do you think so @mts299?

mts299, 2020-09-17:
I think we can close this now? @ecbland and @ksterne with the DAWG authorship PR heading out to the PI's and being most agreed upon.

closed 2020-09-17 by ecbland