SuperDARN/rst #274: Should old-format fit files be limited to radars with 16 beams?

egthomas, 2020-02-19:
I encountered an error when trying to test a loop of

`fitacfserver` > `rtfitacftofit` > `rtfittofitacf`

where the various OldFitRead routines will bail if more than 16 beams are present according to the radar parameter block, eg:

https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.lib/tk/oldfit.1.25/src/fitreadcurrent.c#L186-L187

Likewise, it is not currently possible to perform a circular `fitacftofit` > `fittofitacf` loop using an input fitacf file from a radar with more than 16 beams (`fittofitacf` will exit early and not produce any output, even with the `-vb` flag).

Should this behavior be changed to allow reading of old fit-format files from radars with more than 16 beams?

mts299, 2020-05-25:
**poke** **poke**

mts299, 2020-07-16:
I would say we should make it more than 16 beams if there is still a reason to have **fit** files? 

I imagine we can support it but how much support should we give to the old files? Is this DSWG discussion?

ecbland, 2021-03-15:
This issue has become inactive. I will close it next week if it's still inactive. It has already been added to the [bugs project](https://github.com/SuperDARN/rst/projects/20) so that it won't get lost.