SuperDARN/rst #277: Identified issue for Windows compilation

aburrell, 2020-02-24:
The `setenv` used in time.c is not allowed in Windows and is not good practice because it alters the users environment and does not reset it after running the RST routines.  This is not a good practice because it hijacks the users clock routines. (specficially in `rst/codebase/general/src.lib/time.1.7/src`)

Another potential issue is defining NaN by dividing 0.0/0.0 (I haven't found this in the code, but it was an issue in another area and may be present here so we should look for it.)

ksterne, 2020-02-26:
Yikes!  That doesn't sound that it's setting the OS time.  There is a quick/easy fix for this that you've found or just noting the problem?  I'm wondering if this is something we can get in with v4.4 or if we'll need to make a patch.  For now, I don't think we're "officially" supporting Windows, but that doesn't mean we shouldn't correct this issue.

aburrell, 2020-02-26:
There are two quick/easy fixes.  One (a colleague tested/recommended) is to comment these lines out, apparently this didn't affect the results.  The fix I've implemented previously is to declare a version of these functions if a windows operational system is detected.  You can see what I did here: https://github.com/aburrell/aacgmv2/blob/master/c_aacgmv2/src/rtime.c

egthomas, 2020-02-26:
@aburrell don't these lines at the end of each RST function in `time.c` at least reset the user's time zone to its original value?

https://github.com/SuperDARN/rst/blob/839181fff7f0f3b9a0c9959b5618de8e43e804e0/codebase/general/src.lib/time.1.7/src/time.c#L70-L72

aburrell, 2020-02-26:
@egthomas yes, it looks like it does.  But this will still crash in Windows, because `setenv` and `unsetenv` are not defined.

asreimer, 2020-02-26:
@aburrell what compiler/toolchain are you using in windows?

Also, regarding modification of the user environment: https://stackoverflow.com/a/3416693 so no worries there.

aburrell, 2020-02-26:
I am testing this only using AppVeyor (so no idea).  I don't run anything in a Windows environment, which makes testing tricky.  I am only pointing out an issue a colleague ran into (with their thoughts about it), and a solution I have found to work.

egthomas, 2020-02-26:
This seems like a viable alternative for users trying to compile the RST natively on Windows 10:

https://docs.microsoft.com/en-us/windows/wsl/install-win10

I was just able to compile the master branch using Ubuntu 18.04 in this way.  For fun, here's a screenshot:

[image]

asreimer, 2020-02-26:
@aburrell if you are following the [appveyor.yml](https://github.com/aburrell/apexpy/blob/master/appveyor.yml) you have in apexpy, then you are using mingw. Mingw is a fork of gcc, which is nice because it means building things on windows looks a lot more like building thing on linux/unix.

The solution implemented in [rtime.c](https://github.com/aburrell/aacgmv2/blob/master/c_aacgmv2/src/rtime.c) is a nice one, because it wraps the windows `putenv` in a `setenv` call. I suggest this is done in RST as well.

Running WSL is another option as @egthomas pointed out, but not something all windows users will be willing to do. I have used WSL in the past and really did not enjoy it. Networking does strange things when using a VPN on windows (https://github.com/microsoft/WSL/issues/416) and they do not seem willing to/it is not possible to fix it. We could support the WSL route for users who prefer it, but since the putenv wrapper is such a simple fix I suggest that is also done. This way we don't pigeon hole ourselves.

mts299, 2020-04-29:
Did the suggestion fix the issue?

mts299, 2020-05-25:
@aburrell can this issue be closed?

aburrell, 2020-05-26:
@mts299 it doesn't look like anything was changed, so I believe the issue still exists.

ecbland, 2021-03-15:
This issue has been inactive for more than 12 months. I will close it next week if it's still inactive. I've added it to the [bugs project](https://github.com/SuperDARN/rst/projects/20) so that it won't get lost.

ecbland, 2021-03-30:
Closing this due to inactivity. Please reopen if you would like to work on it.

closed 2021-03-30 by ecbland