SuperDARN/rst #279: map_fit no convergence

mts299, 2020-03-05:
I am trying to process some map files and I keep running into this error for a couple of the years: 

```
No convergence
*** Error in `map_fit': free(): invalid pointer: 0x0000000000b48900 ***
/bin/sh: line 1:  4214 Aborted                 (core dumped) map_fit /home/schmidt/test/19951031.n.model.map > /home/schmidt/test/19951031.n.map
```
There a lot more "no convergence" but I essentially cut it short. Any ideas what is going on? 

I got these errors from 19951031. There might be a bug with freeing up the pointer but I am wondering what the "no convergence" is about.

egthomas, 2020-03-05:
It appears that error message is printed when the singular value decomposition of the velocity matrix in `CnvMapFitVector` fails to converge after 30 iterations.  Maybe there are too few radar or model velocity vectors for that map record?

egthomas, 2020-03-06:
@mts299 if you set the `-vb` option with `map_fit` does that help you narrow down which record is causing problems?

mts299, 2021-05-18:
when I have time will look into this again

closed 2021-05-18 by mts299