SuperDARN/rst #282: Constants values in IDL code

mts299, 2020-03-16:
As @egthomas mentioned and I pointed out in PR #281 that constants I did were just for C code and not IDL. However, there are IDL users. 

If anyone wants to take on this task and make its own PR please do so. I am not as familiar so I am a bit out of my element. 

Some places to look at:
https://github.com/SuperDARN/rst/blob/839181fff7f0f3b9a0c9959b5618de8e43e804e0/codebase/superdarn/src.idl/lib/main.1.25/radar.pro#L800
https://github.com/SuperDARN/rst/blob/839181fff7f0f3b9a0c9959b5618de8e43e804e0/codebase/superdarn/src.idl/lib/main.1.25/radar.pro#L942
https://github.com/SuperDARN/rst/blob/839181fff7f0f3b9a0c9959b5618de8e43e804e0/codebase/superdarn/src.idl/lib/main.1.25/radar.pro#L982
https://github.com/SuperDARN/rst/blob/839181fff7f0f3b9a0c9959b5618de8e43e804e0/codebase/superdarn/src.idl/lib/main.1.25/radar.pro#L995
https://github.com/SuperDARN/rst/blob/839181fff7f0f3b9a0c9959b5618de8e43e804e0/codebase/superdarn/src.idl/app/make_grid.1.0/gtable.pro#L418

mts299, 2020-05-25:
**poke** **poke**

mts299, 2020-07-16:
Anyone want to take this on that knows IDL? @billetd ?

mts299, 2020-11-12:
@egthomas @billetd @ecbland whose willing to take this take on :) all my IDL coders

ecbland, 2020-11-12:
This is not a priority for me at the moment due to time.

mts299, 2020-11-12:
Thanks, I am just trying to knock some issues down :) Not sure how many people use the IDL RST code

billetd, 2020-11-17:
Since IDL has been updated a fair amount since this code was written, basically all relevent constants are now held in the IDL system variable !CONST structure (e.g. !CONST.R_earth). I can test out a branch with these implemented, but it seems pretty straightforward and shouldn't break anything assuming people are using an IDL version from this decade.

closed 2022-01-11 by ecbland