SuperDARN/rst #284: User Errors for erronous dates

mts299, 2020-03-19:
So I am not sure where this error could go, maybe all file producing binaries,, but I just ran into an unusual situation where 19980101 fitacf files had the years: 1998, 1999, 2000, 2008, and 2013 in it.

Furthermore, I went all the way of producing a map file with no errors on this. I know this is potentially "blacklist" the files but at the same time, this unnerving you can have very wrong dates in a file and go as far as making a map file with no errors. I can see this (including myself) a potential headache for users who don't understand what is going on. 

I can make a PR to do some error checking on erroneous dates, just wondering if we have an idea where this error should be put into?

egthomas, 2020-03-25:
For context, this issue was first identified and raised to the DDWG more than two years ago by Simon and myself (and again as recently as 25 July 2019 for a `dat` file with timestamps from 1994, 1995, and 1996 which received no response).  We've continued to raise it at DDWG meetings since then with no apparent action being taken.  In total, there are at least 25,000 `dat` files with timestamp issues which need to be addressed.

I am respectfully (but firmly) in the camp of this being a data quality control / blacklist issue and not something that should be handled in the standard processing software.  There is nothing to stop a user from concatenating together as many `raw` / `fit` / `grid` / `map` files together from as many dates as they like if that suits their data analysis purposes.

mts299, 2020-03-25:
Does that affect the analysis? 
Are jumps in dates accounted for in the analysis? 

Would a warning be appropriate? 

In the previous teleconference, we talked about the error checking of files. I believe this fits the bill.

ecbland, 2020-03-25:
@mts299 Are these errors present in the RAWACF files? If yes, the problem should be handled by the DDWG. If the error arises during RAW->FIT processing then we should investigate. I'm happy to take a look if you can provide a specific example (date + radar)

mts299, 2020-03-25:
19970101 dat files of GBR and SAS is one example. 

My issue is (as mentioned before) DDWG is not perfect at catching all issues, and without having an excellent method to read dat files, it is hard to check these files or know there are issues like how I found cp=0 just a while ago. Non of the checking scripts have that in place to flag the files.

So why not RST?

When processing map files for Kathryn, I ran into many errors/segfaults. However, from dat to rawacf in RST all the way to map data, I got no error/warning about mixed dates. Then even her IDL code did not crash, instead she as a user found some random dates for a known date that didn't match up. A user will come to RST first and ask about this. Of course, it will be pushed to DDWG, but what if it doesn't get handled right away like Evan pointed out this was mentioned 2 years ago. 

If RST could catch some of these well-known errors, then more users than just Evan and Simon could bring it up to the DDWG to help them blacklist the files sooner or at least prevent users from using the files without knowing any better. 

The issue is no working group is perfect and going to catch all problems with software and bugs; that is why we need to help and work together. For example, DSWG needs people to see the issues in our current standards to know which issues to tackle. This is done by having software that allows users to access the data and question things or a distribution that shows standard formats. 

A warning at least would be handy to let the user know of the various date range so they can acknowledge it or investigate it. Again, people can ignore Warnings and Errors.

asreimer, 2020-03-25:
DDWG and DAWG clearly need to be working on this **together**. Punting these problems back and forth doesn't seem to be leading to any progress. Doesn't DDWG use RST as well?

An obvious path forward would be for someone from DAWG and someone from DDWG to build a tool to fix the timestamp issue. The less obvious issue is who has the time.

mts299, 2020-03-25:
@asreimer, I like your idea and was thinking of getting pyDARN to do the error checking for this issue anyways. Also, pyDARN can write in DMAP format so it could fix these issues at the RAWACF level and higher. 

However, it would still be nice to have some other software that produces these files flag the files in case they do slip in so that DDWG can be notified right away. 

I am more than happy to take on this task for RST, I literally made the issue to remind me in the future of the issue.

ecbland, 2020-03-25:
I downloaded the `dat` files from the BAS data mirror for 19970101 for SAS and GBR. To view the contents I used `make_info`, for example
`make_info -old -raw 1997010102t.dat`
All dates are for 19970101.

I then ran `make_fit -old 1997010102t.dat 1997010102t.fit`, followed by `make_info -old -fit 1997010102t.fit`
Again, all dates are for 19970101

I've done this for both radars for the entire day. Can't find a single record where the year is not 1997. 
@mts299 Can you send me one of the files where this occurs?

mts299, 2020-03-25:
What about `make_raw`

egthomas, 2020-03-25:
@ecbland try `1993123122g.dat`

@mts299 `make_raw` only works with `iqdat` files

mts299, 2020-03-25:
sorry, forgot which command I used. Hold on.

ecbland, 2020-03-25:
@egthomas Yes, that `dat` file contains 1 record for 1995 and 9 records for 1996. There are also a few 19940101 records, but that is expected.

mts299, 2021-05-18:
Closing, will look at again if this comes up again

closed 2021-05-18 by mts299