SuperDARN/rst #286: RST vs. Hardware files

mts299, 2020-04-17:
Hey, maybe this needs to be further addressed in a teleconference call, which are we going to have one soon? 

Also, what is the undocumented policy on hardware files with RST? 
Seeing Bill's email today, which makes me feel we need to do an RST patch release to address the changes in the southern hemisphere radars. However, as much as these files don't seem to change, it has happened more frequently, and I am wondering if RST could be separate from hardware file changes? 

Like, make the hardware file repo a submodule to RST as pyDARN does? Thus every time a hardware file changes, the user can still use the same version of RST but update the version of hardware files?

ksterne, 2020-04-17:
Maybe I'm not seeing the difference, but isn't this basically #245?

ecbland, 2020-04-17:
@ksterne It's not quite the same. In fact @mts299 suggested creating a subrepo during a telecon as an alternative to the webhooks solution, but no action has yet been taken. 

Perhaps we should have included this update in the last patch release, but alas we did not.

mts299, 2020-04-17:
.In #245, you asked me to create a PR then merge to `develop,` this still suggests we need to make releases to share the hardware file updates. 
**side note**: we can close that issue if we want to rethink the workflow and focus it here

I can do the subrepo, or try it if you want me too. However, I don't know if we can achieve PR's and having specific hardware files in releases. 
A potentially better solution is to make a SuperDARN hardware repo (as suggested by @egthomas ). Release hardware file changes there via DDWG. Then RST documents the release that was used during testing of the RST release. DSWG might determine we need to add any metadata in the file formats documenting the hardware file release. 
Thus RST releases based on software updates and not hardware updates keeping everything more streamlined or avoid cherry-picking.

The concern with how things are is you get patch releases when in the middle of making a minor release. Some changes like hardware files get forgotten in other patch releases, so then users are strongly encouraged to use `develop` instead of `master`. The `develop` branch does not have a DOI/version number associated with it making reproducibility a potential issue for them. Plus, we want users to use `master`, so we make the changes in `develop` to then be pushed later with a release.

It's like how we don't control the version/release of libraries we use, but we document which version of the library we are using. Sometimes we make a software release when the change in a library affects the software a lot.

mts299, 2020-04-17:
Also, if you want to see how a subrepo works you can see pyDARN `setup.py` and how it integrates with the hardware repo.

mts299, 2020-05-25:
I am going to close this issue now since this issue is now DSWG problem :p 

@ksterne @ecbland  just a poke to make the new repo :p 

Once things are set up I can help with getting hardware files a subrepo and what not :)

closed 2020-05-25 by mts299