SuperDARN/rst #288: Noise removal binary

mts299, 2020-04-27:
# New Feature
New binary for cleaning up noisy data 

## purpose
This binary is to clean up any noisy data to ensure that it doesn't affect higher products. Also, provide a better visual of the sensible Data. Since both Fitacf algorithm has weaknesses in this department, it would be easier to identify interference/noise to remove/smooth it out. 

It is to be post-fitacf as some noise may be useful to the user, and the denoising method(s) are not perfect, so there is some cautionary warning to when using it. 

## considerations
I haven't begun full work into this. Still, I am looking to classify different types of noise/interference using machine learning and then applying various denoising methods to the data to see if specific methods work on others. 

This denoising binary will be a large project as some baselines/statistical analyses will be needed when it comes to determining the best image processing methods.

asreimer, 2020-04-27:
Hey everyone. I appreciate the concerns and the intent behind trying to clean up interference in fitted data that files. Indeed, this is something that we deal with in fitted data from ISRs as well. I have spent the last 10 years digging deeply into the SuperDARN and ISR fitting software and raw data, so I'm drawing from that experience and I'll cite some things as well.

Some context: There's a lot of conversations that have been going on offline about how to deal with this. I'm not sure all of those conversation have been had with DAWG members, so for @mts299 and @kkotyk, apologies, you've heard this before:

1) The specification of this problem isn't well-defined. I appreciate that there is need to deal with "noise", but the definition of what that noise is, what is causing it, and what it's characteristics are hasn't been formalized anywhere. I understand that this is being done to some extent, which is a good first step. I have some thoughts about what this "noise" is, so of which are mixed in to the comments below.

2) Approaching problems from fundamentals and working up from there is my favourite way to solve a problem, so that's what I'm proposing needs to be done. The fundamental issue here is that the noise floor of the radar is increasing, due to interference from man made (HAM band TX-ers) or natural sources (lightning). It would be much better to identify "bad" IQ samples at the voltage level before forming ACFs. Given that almost none of the SuperDARN radars save IQ, the most low level data product that can be used are the ACFs. So, fundamentally, one should be identifying the interference/noise at the ACF level, instead of the fitting level.

3) Given that the data standards working group (? committee?) and the Data Distribution Working Group exist, how does a "noise cleaning" algorithm address their findings/concerns? It seems to me that a project like this has significant overlap with those groups.

4) Because the different fitting algorithms are all different, with different baked in empirical/ad hoc ways of dealing with noise and "badly behaved" ACFs, trying to build a generalized binary for flagging fitted ranges as "noise" is probably going to require a different approach for each fitting algorithm. For example, the chi-squared values for different algorithms are not going to be the same (LMFIT and FITACF flavours of fitters fit different models to the data and the statistical properties of the real/imaginary ACF components are completely different from the magnitude/phase).

5) Try some simple stuff first. Yes, machine learning could probably help, but the investment of time is not immediately obviously worth it compared to doing something easier and simpler. There's a lot of signal processing and data fitting techniques that exist for dealing with this and they don't necessarily involve image processing/classification. For example, the expected behaviour of a noise ACF (an ACF that doesn't have any ionospheric nor ground signal in it) is essentially a delta function. It should have a peak at lag 0, with all the other lags having lower magnitude and random phase progression. What do you expect for fitted parameters when you fit such a thing? Spectral width should be large (approx equal to the 1/(k*tau), where k is radar wavenumber (k=2*pi/lambda) and tau is time difference between lag 1 and lag 0). The Doppler shift should be at the Nyquist limits for LMFIT style fitters (noise ACFs look like very high velocity) and maybe 0? or some crazy value for FITACF style fitters (how do you unwrap the phase of noise? if 0, how does this clash with ground scatter identification? yikes...). chi-squared might help, because it might indicate an overfit (see Chapter 15 "Modeling Data"  of numerical recipes: http://numerical.recipes/).

Numerical recipes basically tells you everything you need to know about how to deal with this issue. Another text that is fantastic is "Understanding Advanced Statistical Methods": https://www.amazon.com/Understanding-Advanced-Statistical-Methods-Chapman/dp/1466512105

ecbland, 2020-04-28:
Before we get into a big debate here, remember that we have a document "Recommended procedure for implementing new algorithms" https://homepage.usask.ca/~pbp672/Data_Analysis_WG_Implementation.pdf

The basic principle is that we are all free to develop new analysis methods, and then approach the DAWG to consider including them in RST after a review process. I realise that we might need to revisit this document now that the Data Standards Working Group exists. 

@mts299 What information were you looking for when you created this issue?

mts299, 2020-04-28:
Figured I could reach out for help, get the data analysis working group interested. Create some discussion on how the working group wants to handle this issue and move forward with hopefully some help. 

Thank you for the document. Not sure where this leaves the issue now... :/

mts299, 2020-04-29:
Closing this issue, will investigate or find someone else for the task. Thank you for the information.

ksterne, 2020-04-29:
Seems as though you forgot to close it?  I think it's OK to leave this open as a capture of a desire of group/PI, though there's certainly lots to think about.  Though this does seem like a bigger picture thing.  It's up to you @mts299 on leaving it open or not.

mts299, 2020-04-29:
woops :p I will close it but if anyone is curious on helping with the problem let me know :)

closed 2020-04-29 by mts299