SuperDARN/rst #290: Zenodo metadata file

ksterne, 2020-05-01:
Somewhat related to #272, it was brought to my attention that Zenodo's not perfect on capturing author names.  It might be helpful in the future to add a `.zenodo.json` file in the root directory to override some of the Zenodo metadata.  One example of this is in:

https://github.com/geodynamics/aspect/blob/master/.zenodo.json

There's other fields that we could more manually control as found in the best practices document from that repo:

https://github.com/geodynamics/best_practices/blob/master/ZenodoBestPractices.md

This should save @egthomas from having to go and edit the metadata for each time where users names aren't populated and/or we decide something different for authorship.  Any general thoughts on this?

asreimer, 2020-05-03:
This is a great idea.

ksterne, 2020-07-08:
We've got a start on this with #305 even though there's more related to it about people ordering and authors/contributors/group members, but that's for another issue.  Thanks for the initial support @asreimer!

closed 2020-07-08 by ksterne