SuperDARN/rst #296: Citing RST documentation

mts299, 2020-05-22:
In pyDARN I created a citing for pyDARN and superDARN data as a reminder to people to do that correctly. I was wondering if we should also make a similar page in RST readthedocs? 

My web dev was actually looking for one on how to cite RST for our website :p

ecbland, 2020-05-22:
The citation for RST is on Zenodo, so would it make sense to just provide the link to Zenodo on readthedocs so we don't have to keep updating it manually with every new release? There is a doi that always points to the most recent version of RST, so we wouldn't need to update anything. 

We should probably resolve #272 first, which I have put on the agenda for the June 1 telecon.

mts299, 2020-05-25:
Fair we can at least leave this issue open for reminding :p 

I was thinking more along the lines of what pyDARN has: https://pydarn.readthedocs.io/en/master/user/citing/

I reference it a lot, and Kathryn has asked why RST doesn't have a page on this. 

It lets the user know how to cite SuperDARN data and you can include the link to the DOI page so people can find it more easily :) 
We can also link to any external libraries we use and their DOI's so that they get credit too. 

Literally our new web programmer was asking me about citing RST haha This is why I bring up the question.

mts299, 2020-07-09:
@ecbland Wondering if we could at least add how to cite SuperDARN data on RST readthedocs and then add it to the website/ 

Kathryn just had to tell someone to cite SuperDARN and include the PI's for specific radars on a paper because they forgot. They used RST as well. 

This might be a bit out of scope but maybe we need to add a warning on output messages with `make_fit` `make_grid`

ecbland, 2020-07-09:
Sure, we can add some info to readthedocs and the website, similar to what you have done for pyDARN. I realise this is a logical place for users to look for this type of information. 

Regarding the PI authorship rules, this is definitely not DAWG's problem. I'm happy to help communicate the information on readthedocs, but the PIs should provide the text. Can you please ask Kathryn?

mts299, 2020-07-16:
As the temporary "for now" fix, pyDARN and our website: https://superdarn.ca/data-download
document how to cite our data. We are doing this as due-diligence while the PIs come up with a better how to cite. 

Kathryn has "okay" for our website so I can imagine this could be on RST as well until something better comes along. 
Thoughts?

ecbland, 2020-07-16:
@mts299 That's helpful, thanks! We can add this to the software page on the website and to RST readthedocs

mts299, 2020-09-17:
@ecbland have we done this? I can do this in the next few days.

ecbland, 2020-09-17:
@mts299 Nothing has been done about this yet, but I guess it's a simple copy-paste job from the pyDARN documentation or the SuperDARN Canada website. That would be great if you can get the ball rolling...thanks.

ecbland, 2020-09-19:
Closing this now that #342 has been merged. Thanks @mts299 :smile:

closed 2020-09-24 by ecbland