SuperDARN/rst #300: Guidelines... rules... responsibilities?

mts299, 2020-06-03:
Sorry, I didn't know how to classify this, but what is the rule/guideline on deleting branches? 

Noticed we have 21 branches in RST and I cannot imagine all of them are still needed :p

ecbland, 2020-06-04:
Thanks for noticing this, @mts299. The short answer is that we don't have any guidelines for this. This is something we should take a look at. 

There are a few branches that might need to be kept. I'm thinking specifically about the hardware file modifications that @egthomas made for the Syowa radars (#20), which never got merged because we could not confirm the information in them. I know this is not the DAWG's responsibility, but the conversation & updates should be preserved until someone resolves the issue. We could just make an issue for DSWG with all the relevant info and then remove it from here. 

I'll assign this to the chairs -- we'll take a look soon.

mts299, 2020-06-04:
Sounds good! and don't delete `elevation library` branch just yet :p 

I can delete some of my very few merged PRs if that helps.

egthomas, 2020-06-04:
Does closing a pull request still protect the branch like merging does?

mts299, 2020-06-04:
Sorry getting confused on the lingo with closing, closing a PR keeps the branch alive. PRs are just GUI for merging code essentially (obviously, they do a lot more). However, if you `delete` a branch from the branch *tab8 and it is not merged through a PR then it is lost. 

So I think the answer is no @egthomas if I understand your question correctly.

mts299, 2020-06-04:
I think for branches that are not in PRs or never have been PR'd like if I made one and forgot about it... then we should discuss in a teleconference to make sure that person is not using it. Same with branches that were in a `closed` PR. Like my `elevation library` branch.  

After releases, we probably can just `delete` previously merged branches.

egthomas, 2020-06-04:
@mts299 yes that's exactly what I meant, thanks.

ecbland, 2020-06-05:
@egthomas Shall we create an issue for DSWG regarding the Syowa hardware files? Even if we kept the `syowa_hdw_fix` branch in RST, I'm worried this issue will just get buried. 

One option is to upload your proposed changes to the hardware files in a new issue on `dswg-published-docs` and link to the conversation about them in #20. Then it would be safe to delete the branch.

mts299, 2021-01-20:
@ecbland should we discuss this at the telecon meeting, saw we have 27 branches. I deleted some of mine. 

Just looking at issues to focus on.

mts299, 2021-02-09:
Looks like we can try to remember to do this every so often with meetings.

closed 2021-02-09 by mts299