SuperDARN/rst #302: FOV bug in grid_plot

ninaer, 2020-06-08:
When trying to plot fov in a grid plot using:
`grid_plot -fov somedate.north.grd`
All radar field of views are plotted, and not just the ones that contains data. This error message also occurs in the terminal:

> ERROR: coefficients are not valid for altitudes above 2000 km: 2016.378259.
       You must either use field-line tracing (TRACE or ALLOWTRACE) or
       indicate that you know this is a very bad idea (BADIDEA)

I use rst version 4.3.2 and ubuntu version 18.04.4 LTS

mts299, 2020-07-16:
@aburrell @billetd @ecbland @egthomas anyone know about this?

egthomas, 2020-08-06:
Sorry, this functionality was never added to `grid_plot` - it was only added to `map_plot` via the `-gfov` option.  It probably wouldn't be difficult to port to `grid_plot`.

ecbland, 2020-08-13:
Closing this now that #319 has been merged. The new functionality has been added to `grid_plot`, but we have not addressed the error message. I will open a new issue for that. Thanks @egthomas for implementing this new feature!

closed 2020-08-13 by ecbland