SuperDARN/rst #303: Dat to RAWACF issue?

mts299, 2020-06-15:
Once again I am regenerating rawacf -> fitacf 2.5 -> map files for Kathryn/SuperDARN community and as I finish the more recent years, I am wondering if my dat to rawacf files are safe to convert to fitacf 2.5/map files? 

@egthomas in the DDWG meeting you mentioned a potential issue with this conversion. Could you please elaborate more? 
If this is an issue/bug can we make it a high priority to fix as well?

ecbland, 2020-06-15:
Just to add to this, @kevinkrieger reached out to me & @ksterne last Friday asking if the DAWG can help develop a tool to check dat files and the dat->rawacf conversion. We haven't discussed this properly yet, but this might be a good opportunity for DAWG & DDWG to work together :)

mts299, 2020-06-15:
I would be more than willing to help out as we will need to verify these files to be able to put them in an archive on FRDR which the PI's approved. One thing to consider if this would be an RST task or should we look at pyDARN to help with this?

mts299, 2020-07-08:
@egthomas comment?

ksterne, 2020-07-08:
Hey @mts299, I know @egthomas is out for the better part of the next month (I think) so I wouldn't expect a comment quickly.  There's some errors I've noticed on `dattorawacf` but I haven't been able to capture it.  I can try to expand on it sometime in the future, but not today.

mts299, 2020-07-08:
Okay my bad so his activity the other day and week so thought he was back.

egthomas, 2020-07-10:
Sorry, like @ksterne said I'm currently out of the office on leave and was just trying to help with the release a few days ago.

I don't have any concerns with the current state of the `dattorawacf` conversion process in terms of software.  The issue I raised during one of the WG meetings (can't remember which one) was probably related to the data mirror / distribution aspect and keeping the converted rawacf files in separate directories from the original rawacf files produced on-site.

There may be an inconsistency in how we handle the `origin.code` parameter when executing `make_raw` or `make_fit` compared to `dattorawacf` or `fittofitacf`, but that is probably out of scope for this particular issue.

mts299, 2020-07-16:
Sounds good I will close this issue and pass it onto DDWG for checking files made by `dattorawacf` to determine if need to be regenerated.

closed 2020-07-16 by mts299