SuperDARN/rst #306: Transition away from RFC documentation/files

ksterne, 2020-07-06:
Something to capture for our next release that's come up in #297, is completely transitioning info out of the older RFC's since we've got a new list for documentation going.  In particular, making sure the gridded vector means and medians are labeled correctly, but I believe these were fixed in the new documentation.

For reference, the RFC's are in:
rst/codebase/superdarn/src.doc/rfc/

and the new documentation is in (or the info should probably move to):
rst/docs/references/general/

ksterne, 2020-10-26:
I know this is pretty far in development, but does anyone thing they'd have a chance to knock this one out.  I might be able to find time, but it wouldn't be until next week at the earliest.

mts299, 2020-10-27:
What is needed for this?

ksterne, 2020-11-02:
I think it's mostly just making sure all of the info in the RFC directory is captured in the new directory mentioned in the original post before removing the RFC files/directory.  I can see if I can tackle that this week unless someone else would like to.

ksterne, 2020-11-11:
Of course, "this week" really meant a different week. Going to try to tackle this tonight and tomorrow.

ksterne, 2020-11-20:
Hey all, in working on this I noticed there's a difference between the RFC and the readthedocs info on the rawacf file.  Just trying to short-cut from going back and looking at that pull request, but does anyone remember about the mxpwr parameter's units?  In the readthedocs part (for now only in docs), it's labeled as units of dB whereas in the [RFC](https://superdarn.github.io/rst/superdarn/src.doc/rfc/0008.html), it has units of kHz.  Anyone remember about this?

ecbland, 2020-11-20:
kHz makes no sense as a unit of power. I think it should actually be kW, same as the `txpow` parameter. 
It looks like the dB for `mxpow` was added when we wrote the `fitacf` format documentation (and then copied over to the `rawacf` documentation). The data type is integer, so it seems unlikely that the units would be dB.

egthomas, 2020-11-20:
Not sure whether they're still accurate, but the comments in [raw.h](https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.lib/tk/oldraw.1.16/src/raw.h) of the `oldraw` library are copied nearly verbatim from the original `radops.h` file (used by the RADOPS 386 software circa-1993), and state:

```
	int32 MXPWR,	/*	Maximum power allowed (see note 3) */

	3.During the gain setting routine, the system will attempt to add
	  enough attenuation so that the maximum reflected power is less than
	  MXPWR.  If this is not possible, the system will set the error code
	  (ERCOD) to indicate the receiver overload condition.
	
	  During the clear frequency search, the system will find the clearest
 	  frequency in the range specified.  The Noise level determined for
	  that frequency will be stored in the parameter NOISE.  If NOISE is
	  greater than LVMAX, the error code will be set to indicate the
	  no clear frequency condition.
```

ksterne, 2021-01-13:
Sorry for the slow progress on this, holidays and a week long trip to Wallops slowed things down (plus getting into other work).  I've fairly well gotten this finished off and looking to do a PR here soon.  One thing before I do, I'm not quite sure what to do with the [ROS related documentation](https://superdarn.github.io/rst/superdarn/src.doc/rfc/0026.html).  I've started a legacy software section in the table of contents, so maybe this could go under that since the RST included ROS code during the RST/ROS 3.x versions?

Are there other thoughts on where this could go (other WGs)?  It's certainly not the latest and greatest, but does cover a few radars (as well as could be abstracted to many other SuperDARN radars).

mts299, 2021-01-14:
@ksterne No worries, take your time, I don't think there is a lot of pressure. 

I asked my group about where ROS docs could go seeing how it's not exactly a DAWG thing but we don't really have the OSWG... maybe DSWG with that radar audit issue?

mts299, 2021-01-14:
What about here:  https://github.com/SuperDARN/ros-archive/?

ksterne, 2021-01-14:
Certainly agree a good area would be an operating software area, but that WG isn't around anymore.  I think the ROS-archive repo is a good idea.  I'll work to see if and/or how that can work.  Again, this is getting close to being completed.

I'm thinking of keeping both the readthedocs info but as well putting a deprecation notice on this before it's completely removed.  Or does this seem overly cautious and I should do a "rip-the-bandaid" by removing the RFC info altogether?  I'd like to go the deprecation route as a habit or example of how to do this in this repo, but I know it's not working code (which makes it OK if we mess up the process).  Any thoughts?

closed 2021-02-22 by mts299