SuperDARN/rst #307: Zenodo error:  "The metadata file is not valid JSON."

egthomas, 2020-07-07:
The version 4.4 release failed on Zenodo because of an error with the metadata file.  Using jsonlint.com to validate the `.zenodo.json` file, it looks like there is an error here:

```
Error: Parse error on line 36:
... "Schmidt, M.T."			"orcid": "0000-0002-
----------------------^
Expecting 'EOF', '}', ':', ',', ']', got 'STRING'
```

ksterne, 2020-07-07:
Ah, fiddle-sticks!  Let me see if I can correct that real quick

ksterne, 2020-07-07:
Or I guess @egthomas is doing it.  I don't know how we missed the comma. Whoops.

ksterne, 2020-07-07:
Looks like we have a Zenodo release, so things are good here @egthomas? I'll hold off sending out an announcement until you're good and done with edits.  Feel free to close this when you're done.

egthomas, 2020-07-07:
Yeah everything looks fine now so this issue can be closed.

However while we're on the subject, has there been any decision on the author list for 4.3.1 and 4.3.2?  Currently [4.3.2](https://zenodo.org/record/3775981) uses the "Project Members" category rather than "Authors" since we had discussed trying that during the WG meeting.

ksterne, 2020-07-07:
That works. I guess I'll (or someone) need to look into how to configure the `.zenodo.json` file to follow that convention.  I'd forgotten we'd done that since it all looks about the same on my end on Zenodo. Sorry! I don't think it's too big of a deal for either of those.

closed 2020-07-07 by ksterne