SuperDARN/rst #310: BUG: compilation failed at stdkey.1.5

aburrell, 2020-07-09:
Just went to upgrade to the last release of RST on my processing system.  This required updating to the latest version of CDF (3.8), and making a change to one of the makefiles:

In `$RSTPATH/codebase/general/src.lib/stdkey.1.5/src/makefile` `SLIBS` and `LIBS` are both equal to `-lm`, which results in the following command that errors:
```
================================================================================
Compiling Library:$RSTPATH/codebase/general/src.lib/stdkey.1.5
================================================================================
make clean
rm -f *.o
make
rm -fv *.o
rm -fv paltable.c
rm -fv make_palette
removed `make_palette'
cc -o make_palette make_palette.c -I$RSTPATH/include/base -lm
./make_palette
making spectra.
making color bars.
printing palette table.
printing rgb tables.
make.version $RSTPATH/codebase/general/src.lib/stdkey.1.5/src/..
make.hdr $RSTPATH/codebase/general/src.lib/stdkey.1.5/src/../include $RSTPATH/include/general 
$RSTPATH/codebase/general/src.lib/stdkey.1.5/src/../include/stdkey.h
`$RSTPATH/codebase/general/src.lib/stdkey.1.5/src/../include/stdkey.h' -> `$RSTPATH/include/general/stdkey.h'
cc -fPIC -Wall -O3 -ansi -D_GNU_SOURCE -D_LINUX -I$RSTPATH/include/base -I$RSTPATH/include/general -c paltable.c    
mkdir -p $RSTPATH/lib
ar -r $RSTPATH/lib/libstdkey.1.5.a paltable.o
ld -shared -lm -lz  -o /$RSTPATH/lib/libstdkey.1.5.so \
              paltable.o -Bstatic -L$RSTPATH/lib -lm -lm
ld: cannot find -lm
make: *** [stdkey] Error 1
Compilation Aborted.
```

Commenting out those two environment variables results in this command that succeeds:
```
================================================================================
Compiling Library:$RSTPATH/codebase/general/src.lib/stdkey.1.5
================================================================================
make clean
rm -f *.o
make
rm -fv *.o
rm -fv paltable.c
removed `paltable.c'
rm -fv make_palette
removed `make_palette'
cc -o make_palette make_palette.c -I$RSTPATH/include/base -lm
./make_palette
making spectra.
making color bars.
printing palette table.
printing rgb tables.
make.version $RSTPATH/codebase/general/src.lib/stdkey.1.5/src/..
make.hdr $RSTPATH/codebase/general/src.lib/stdkey.1.5/src/../include $RSTPATH/include/general 
$RSTPATH/codebase/general/src.lib/stdkey.1.5/src/../include/stdkey.h
`$RSTPATH/codebase/general/src.lib/stdkey.1.5/src/../include/stdkey.h' -> `$RSTPATH/include/general/stdkey.h'
cc -fPIC -Wall -O3 -ansi -D_GNU_SOURCE -D_LINUX -I$RSTPATH/include/base -I$RSTPATH/include/general -c paltable.c    
mkdir -p $RSTPATH/lib
ar -r $RSTPATH/lib/libstdkey.1.5.a paltable.o
ld -shared -lm -lz  -o $RSTPATH/lib/libstdkey.1.5.so \
              paltable.o -Bstatic -L$RSTPATH/lib  
cd $RSTPATH/lib; ln -sf libstdkey.1.5.a \
               libstdkey."1".a 
cd $RSTPATH/lib; ln -sf libstdkey.1.5.so \
               libstdkey."1".so 
rm -fv paltable.c
removed `paltable.c'
mv -f stdkey.h $RSTPATH/include/general/stdkey.h
```

It seems like including the `-lm` flag three times on this command was not helpful and should probably be changed.

OS: CentOS release 6.10 (Final)
CC: gcc version 4.4.7 20120313 (Red Hat 4.4.7-23) (GCC)

ecbland, 2020-07-15:
Thanks for the bug report @aburrell! You said you upgraded to CDF3.8. Do you think that is related to this bug?

aburrell, 2020-07-15:
No, I don't.  I don't think the command line that failed is at all related to CDF3.8.  It could be that the compilers previously used to test this allowed bad input and the compiler on this machine doesn't, but I am not very compiler savvy.

ecbland, 2020-07-15:
Ok, thanks for the clarification. Perhaps @egthomas knows something about the CDF issue? I am also not compiler-savvy so I'll defer to the experts about the compilation fail (@mts299?)

egthomas, 2020-08-06:
It looks like the error with the `stdkey` library makefile was just recently introduced in https://github.com/SuperDARN/rst/commit/847280646ad6ea558a4bf909417661e81536934e (#281).

aburrell, 2020-08-06:
Well spotted @egthomas

egthomas, 2020-08-07:
Does @aburrell's fix warrant a patch release since there have already been another user reporting compilation failure on v4.4?

aburrell, 2020-08-10:
I would say yes.  I can make the PR, if you like.  Do I just make a branch from master?  Confirming to ensure I follow proper procedure.  Otherwise, tag me as a reviewer and I'll test it.

egthomas, 2020-08-10:
@aburrell since this compilation error does not seem to occur on Ubuntu systems and that is all I have easy access to right now, maybe you should go ahead and create the pull request.  My understanding is that the pull request should be made directly to master to avoid copying over any recent modifications to develop, but @ksterne or @ecbland (or someone else) should probably verify that procedure.

ksterne, 2020-08-10:
I agree that we should do a patch release for this.  I'd like to check with @mts299 though if there's some reason that those lines were added in.  Marina, do you know if these lines are needed:

ecbland, 2020-08-10:
> @ksterne one day we'll put together the release checklist

I made a draft release checklist over in the `dawg` repo (see the release guidelines PR). There are no details about how to create a patch release, but that should be added.

ksterne, 2020-08-10:
My apologies @ecbland!  I'd forgotten you'd started working on that.

mts299, 2020-08-10:
Maybe we should do what is called a "HotFix" where you push the fix to master with no release number to it as it is small change. If we need a change for zenodo then a patch number could be changed. This will preserve other PR's like #293 to be introduced in a larger release.

ksterne, 2020-08-13:
Hey @mts299, sorry on a slower reply.  I think it's true we could do just a hotfix to the master branch.  But Zenodo will pick that up and create a new citation for it.  So then there would be confusion as there would be 2 citations for version 4.4.  I saw this with version 4.3.2 (#287).  So I think it's easiest to add this as a patch version.

Did you see my question on if you remember what the lines you added were doing?  If you don't remember, that's OK! Just want to make sure there's some reason I (and others) aren't understanding why these lines are there.

mts299, 2020-08-13:
Actually (if the same as pyDARN) when you push to master nothing will happen on zenodo, its when you create a release that zenodo will pick it up and DOI it. 
So the idea would be to push to master then tag it with v4.4.1 then announce the bug fix. 
That way it doesn't encapsulate the compiler change, the other changes we made. 

Also with it being a patch and only testing that one bug, you shouldn't have to go through the long testing period with releases.

Hopefully, this clears things up @ksterne

mts299, 2020-08-13:
I would also add we should avoid tagging PR's and commits that caused the bug as these can associate with people and lay blame when the real issue is what lines of code were changed to fix the problem. I have not looked at this problem since everyone was on it, and no links to code lines were presented, so I didn't want to go digging through a PR/commit number seeing others have already done that. 

I can take a look Monday as I am away tomorrow and have meetings tomorrow.

mts299, 2020-08-13:
Okay, so I am looking at it now because of restructuring. I will see if I can fix it today.

ksterne, 2020-08-13:
I think you are right on the Zenodo capture as it only goes on the new tags.  Thanks for reminding me @mts299!  (I had more on the branching/PR to setup for this, but looks like you read my keystrokes!)

My apologies on pulling out specific commits here.  My only thought was to try to make it easier for you to go back to what was going on.  I was also calling on you as at least I (maybe others) don't know enough about the compiler options to make a good evaluation/determine.  So your knowledge on this is very helpful.  I can understand how this can be viewed negatively and will limit this in the future.  I don't think it was at all something wrong that happened, a lot of us tested the PR at the time and never found it.  So, if anything the "blame" (again, I wasn't trying to blame) is on us for not finding this bug during testing.  It's all not a problem though as long as we can fix it (which we can!).

aburrell, 2020-08-14:
Thanks for making PR #323 to address this issue @mts299 .  Seems like this was the best way to do things, since apparently I'm the only person on DAWG who can test the bug :-P.

mts299, 2020-09-14:
Fixed bug in RST 4.4.1 release

closed 2020-09-14 by mts299