SuperDARN/rst #314: Definning Error Codes and messages

mts299, 2020-07-16:
As we move to better practices and something more user/developer-friendly for the Scientific community. We agreed on introducing error codes #240 into RST but nothing has been defined. 

`errno.h` a C library defines a bunch of standard error codes that we can use as well as define our own in a developer guideline.

ecbland, 2020-07-21:
@mts299 What are you asking/proposing in this issue?

mts299, 2020-07-23:
It's more a place holder for the discussion around error codes. 

1. are we going to use errno.h to define the more mundane error and use it as a future development tool? 

2. What error codes can we define for RST specific errors? This will probably have to happen a bit later as we dig into some code. 

Really I made this a reminder to write documentation on error codes for RST :p

mts299, 2021-01-20:
Gonna close this and make some smaller tasks on the project to then make progress while I have time.

closed 2021-01-20 by mts299