SuperDARN/rst #316: Having issues with rst installation and linking with IDL

jameslane26, 2020-08-06:
Hi,

I've been following the installation instructions provided here: https://radar-software-toolkit-rst.readthedocs.io/en/latest/user_guide/mac_install/

Firstly, for some reason, whenever I download the ZIP, or use git clone etc., the '.profile' folder is not downloading with everything else.

I have two separate file directories - one is my IDL one (/exelis/idl85/....) which contains an idl_setup.bash script - is this the bash file I need to edit to include the RSTPATH/.profile.bash file when I can get it? The other file directory is the rst one which contains the folders /bin, /build etc. but not the /.profile file.

Also, I should mention that I do have what I think is a somewhat working version of rst, whereby if I type '. .bashrc' at the command line, I can then use the functions (e.g. map_fit, etc.), but I don't think this is linked to my IDL in any way?  Apologies if this seems vague - this was installed for me at the beginning of my PhD and it's only almost 2 years later that I'm now coming to use it.

Overall, what I want to be able to do is write some IDL code to read in e.g. FITACF files and then be able to plot the data and customise the plots myself, since currently I'm restricted to using the command line functions like time_plot etc. only.

Thanks,

James

aburrell, 2020-08-06:
Hi James, simple question for context: how are you viewing the RST files you download?  Is it through a finder window or a terminal window?

jameslane26, 2020-08-06:
Hi, I've mainly been using a finder window.

aburrell, 2020-08-06:
Ah, this is your problem.  The finder window hides all hidden files and directories (those whose names start with a `.`), like `.profile`.  If you open a terminal window and:
```
cd $RSTPATH
ls -a
```
You'll see the "missing" directory and files.

jameslane26, 2020-08-06:
Ahh yes I can see that now, thanks!

closed 2020-08-06 by jameslane26