SuperDARN/rst #317: RST installation on MacOS Catalina

mtwalach, 2020-08-06:
Hi everyone,

I am trying to install RST on Catalina for the first time and am having some trouble: 

1) On running make.code I get the following error (which is new to me):

================================================================================
Compiling Library:/Users/maria/rst/rst/codebase/base/src.lib/graphic/xwin.1.5
================================================================================
make clean
rm -f *.o
make
make.version /Users/maria/rst/rst/codebase/base/src.lib/graphic/xwin.1.5/src/..
make.hdr /Users/maria/rst/rst/codebase/base/src.lib/graphic/xwin.1.5/src/../include /Users/maria/rst/rst/include/base 
/Users/maria/rst/rst/codebase/base/src.lib/graphic/xwin.1.5/src/../include/xwin.h
/Users/maria/rst/rst/codebase/base/src.lib/graphic/xwin.1.5/src/../include/xwin.h -> /Users/maria/rst/rst/include/base/xwin.h
cc -Wall -O3 -ansi -D_GNU_SOURCE -D_DARWIN -I/opt/local/include -I/Users/maria/rst/rst/include/base -I/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/Tk.framework/Versions/8.5/Headers/X11/include   -c xwin.c    
xwin.c:35:10: fatal error: 'X11/Xos.h' file not found
#include <X11/Xos.h>
         ^~~~~~~~~~~
1 error generated.
make: *** [xwin.o] Error 1
Compilation Aborted.

The problem seems to be that it can't find the Xos.h file which I can't either! It's not in the X11 folder and when I run `locate Xos.h` nothing is found. Anyone else had any similar problems? Or knows how to overcome this? 

2) Also, another error I have had (this time around and in the past when installing RST on Mac in general) is that the folder structure which RST expects ".../X11/include/X11/xlib.h" actually just seems to be ".../X11/xlib.h" and every time I have to find the right place to put a fake folder with the right symlink in. Can we change whatever code is needed for future versions of RST and take out the "include"? Or is that a simplistic view? I don't think I am the only one but I am curious: has anyone with a Mac not had this issue? 

Thanks,
Maria

aburrell, 2020-08-10:
There are so many problems with Catalina that we aren't allowed to upgrade yet.  For Mojave, the X11 directory structure is the same as what RST currently expects (`/usr/X11/include/X11/Xlib.h`).  If the Catalina directory structure is very different, we should consider and discuss restructuring the X11 environment variable.  I agree that requiring users to make soft links isn't great.

mtwalach, 2020-08-10:
Thanks Angeline. It's a new system and I am trying not to downgrade :(

aburrell, 2020-08-10:
Yeah, I get that :( Also, we'll all have to upgrade eventually and I doubt Mac will fix things to make them easier on us users.  Did I interpret your text correctly if I say that your problem is fixed if you make a symlink?  If so, I think we should bring up the X11 environment variable definition in the next DAWG meeting.

mtwalach, 2020-08-10:
Problem 2) is fixed, but I am still struggling with 1) - I think it's something to do with XCode changing because I can't find the Xos.h file and I don't know where I get it from.

mts299, 2020-09-03:
Where are we at on this? did 4.4.1 fix it? @billetd do you have any expertise in this since you got a Mac?

billetd, 2020-09-03:
I am on Catalina, and I recall having problems with XCode similar to what @aburrell mentioned. I think it involved me redirecting some lines make.code to the correct directories.

mtwalach, 2020-09-04:
I've still not had any luck - Does anyone know in which folder I would usually find the `Xos.h` file? As I said, the suggestion @aburrell made fixed problem 2.), but not my main problem 1.). None of the folders that have anything to do with X11 have any file named `Xos.h` in them. Does anyone know what this file does/is? I'm guessing it's some kind of library but how do I get it if XCode doesn't include it as it usually would?

mtwalach, 2020-10-13:
Okay, I have fixed the above: 
It turns out that if you install Xcode and all the dependencies listed in the instructions using "homebrew", you miss out on this very important library file. The way to fix it, is to install XQuartz which puts all the relevant X11 libraries into /opt/X11/ etc 

I have since run into another problem, but I guess I'll open up another issue.

closed 2020-10-13 by mtwalach