SuperDARN/rst #318: RST build error on linux

alexchartier, 2020-08-06:
I followed the instructions to build on a linux machine, but got an error from make.code that I don't understand. It seems to be caused by the double -lm at the end of the libstdkey.1.5.so section. 

make.hdr /homes/chartat1/rst/codebase/general/src.lib/stdkey.1.5/src/../include /homes/chartat1/rst/include/general 
/homes/chartat1/rst/codebase/general/src.lib/stdkey.1.5/src/../include/stdkey.h
`/homes/chartat1/rst/codebase/general/src.lib/stdkey.1.5/src/../include/stdkey.h' -> `/homes/chartat1/rst/include/general/stdkey.h'
cc -fPIC -Wall -O3 -ansi -D_GNU_SOURCE -D_LINUX -I/homes/chartat1/rst/include/base -I/homes/chartat1/rst/include/general -c paltable.c    
mkdir -p /homes/chartat1/rst/lib
ar -r /homes/chartat1/rst/lib/libstdkey.1.5.a paltable.o
ar: creating /homes/chartat1/rst/lib/libstdkey.1.5.a
ld -shared -lm -lz  -o /homes/chartat1/rst/lib/libstdkey.1.5.so \
              paltable.o -Bstatic -L/homes/chartat1/rst/lib -lm -lm
ld: cannot find -lm
make: *** [stdkey] Error 1
Compilation Aborted.

aburrell, 2020-08-06:
I found and reported this issue here: #310

alexchartier, 2020-08-06:
Thanks Angeline - that seems to have fixed it for me.

closed 2020-08-06 by alexchartier