SuperDARN/rst #322: AACGM_v2 error: "coefficients are not valid for altitudes above 2000 km"

ecbland, 2020-08-13:
The `grid_plot` routine sometimes returns the following error from AACGM_v2:

```
grid_plot -x -def -fov [grid_file]

ERROR: coefficients are not valid for altitudes above 2000 km: 2191.461117.
       You must either use field-line tracing (TRACE or ALLOWTRACE) or
       indicate that you know this is a very bad idea (BADIDEA)
```

In the above example, the error occurs when the `-fov` option is included. The routine still works and the plots look ok. 

This might happen with other RST routines as well - please add them in the comments if you find them!

See also #302

egthomas, 2020-08-13:
This appears to happen for the SAS, PGR, INV, RKN radars only - and I think it's because their max range is defined as 225 gates in their hdw.dat files, which is used by `grid_plot`'s `make_fov` in the call to `RPosMag`, eg

https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.bin/tk/plot/grid_plot.1.12/make_fov.c#L65)

`field_plot`, `grid_plot`, and `map_plot` all appear to use `site->maxrange` from the hdw files when calculating the radar FOVs.  One solution could be to set an upper allowable limit on the for loops using `site->maxrange` to something like 110 range gates.

aburrell, 2020-08-14:
The simplest way to change this is to default to using `ALLOWTRACE`.  As defined in aacgmlib_v2.h, `ALLOWTRACE` works by: "if height is >2000 km use tracing, else use coefficients".

egthomas, 2020-08-14:
My only concern is that the FOVs drawn by those routines are possibly meant to be more representative than wholly accurate, as they already use hardcoded "standard" values of `rsep` (45 km) and `frang` (180 km).

Or looking at it from another perspective, do the Canadian radars regularly use 225 range gates or is that more of a theoretical maximum?  So should the software draw a FOV with all 225 ranges all the time?  For example, I know we have used 200+ range gates when running 15 km (or less) range resolution modes at the CV radars but have kept a maximum range of 110 in the hardware file as that's what we use 99% of the time.

aburrell, 2020-08-14:
> Or looking at it from another perspective, do the Canadian radars regularly use 225 range gates or is that more of a theoretical maximum?

From what I have seen, a maximum of 100 is typical for INV and CLY.

asreimer, 2020-08-14:
There are experiments running 15 km range gates at INV and CLY which is why
the hdw files have 225 for max range. Definitely infrequent, but they exist.

egthomas, 2020-08-14:
Right, so using 75 range gates at 45 km resolution should result in the same size FOV as using 225 range gates at 15 km resolution.  I guess the real question is, do we want to use the actual radar operating parameters when calculating the outline of each FOV for these plotting routines, or is the standard 75 range gates at 45 km sufficient to provide a representative FOV if applied to all radars?

At a minimum, I think we can probably all agree the software should not be drawing FOVs with 225 range gates using the hardcoded `rsep` of 45 km.

ecbland, 2020-08-14:
Thanks for the input everyone! 

I think it makes sense to just set an upper limit on the number of range gates. I'll also add that `site->maxrange` in the hardware files is really only intended for [memory allocation](https://github.com/SuperDARN/rst/blob/5a4f88df543ae63c6771cfbb8616c5b6f2893367/tables/superdarn/hdw/hdw.dat.rkn#L83).

I noticed that the number of gates is fixed at 75 in [`fov_plot`](https://github.com/SuperDARN/rst/commit/5442f4f502c0423f02cf49cd5d78a8de016440b9) (possibly to solve the same problem?) However, it might make more sense to set an upper limit of 110 gates since some radars really do operate with larger fields of view than the standard ~3500km (75 gates). 

Whatever we decide, the field of view plotting should be made consistent across all of RST. 

> @egthomas: do we want to use the actual radar operating parameters when calculating the outline of each FOV for these plotting routines...?

This would be nice, but I don't think this information is contained in the grid files 😞

asreimer, 2020-08-14:
Really, the point of gfov_plot is to just roughly visualize the FOV of the radars right?

If that's the intent, then generating the FOV assuming only 75 range gates with 45km rsep seems most reasonable.

egthomas, 2020-08-17:
> This would be nice, but I don't think this information is contained in the grid files

@ecbland good point!  I had forgotten that information is lost with the fit to grid conversion.

closed 2020-09-09 by egthomas