SuperDARN/rst #325: Removing versions

mts299, 2020-08-24:
We brought this up 2 years ago and something that might help de-clutter rst for easier moving around I was wondering if we can discuss removing version numbers in our version repository? 

I believe I tested in PR #169 on `make_fitex`.

mtwalach, 2020-08-25:
Just wanted to check: I think by this you mean getting rid of all the subfolders for the routines that have older versions as well as new ones? As opposed to getting rid of `rst/.rst.version` and the functionality of `--version`?

mts299, 2020-08-25:
@mtwalach Thank you for clearing that up. You are correct, I mean subfolders in RST not version of rst or versions of algorithms people use daily like FITACF 2.5 and 3.0

mts299, 2020-09-03:
@ecbland @ksterne Shall I start some draft PR's on removing versioned ... versioned subfolders? 

Then we can discuss further.

ecbland, 2020-09-03:
@mts299 Yes please!

Everyone: if you have any general questions/concerns about this update then please speak up now 😄

egthomas, 2020-09-03:
Will there still be hundreds of duplicate commits for each directory which has its version number removed?

mts299, 2020-09-03:
probably not sure if they are duplicates or rewrite history.

mts299, 2020-10-05:
Noticing there have been some `git mv` in other PR work. I made a PR on using `git mv` for the version numbers #336 compared to #333

mts299, 2020-10-05:
Let me know which one we prefer both have caveats

mts299, 2021-01-29:
Wondering when we do this if we can also get rid of the `tk` folder that only seems to be in the `superdarn` folder and seems to have no relevance. 

I find it annoying while coding.

mts299, 2021-05-18:
Will do later

closed 2021-05-18 by mts299