SuperDARN/rst #329: BZIP2 Support

mts299, 2020-09-03:
So I know we use to do gzip support wondering about bzp2 support? 

I just find is annoying downloading files, unzipping then using RST and then I have large unzip files on my desktop. 

Maybe this is me being a lazy user :p

ecbland, 2020-09-03:
@mts299 Would you mind providing a little more detail on what you're suggesting here? Do you mean that I could unzip and process a rawacf file with a single command, i.e. `make_fit -bz2 [filename].rawacf.bz2 > [filename].fitacf`?

Do we still have gzip support? If not, do you know why it was removed?

mts299, 2020-09-03:
Yes! that is what I mean or `make_grid -c <filename>.fitacf.bz2 > <filename>.grid`

I will look to the lit to see where it was provided. I just remember someone saying you could use gzip files in  RST.

mts299, 2020-09-03:
I thought it was a thing but I cannot finid it in the code or older record information maybe @egthomas might know more?

egthomas, 2020-09-03:
We looked into this several years ago and there were no discernible improvements in processing time when working with a bzip2 file directly compared to first uncompressing and then calling whatever RST binary. Of course that may no longer be true, but at the time we decided it was not worth pursuing.

mts299, 2020-09-03:
I might pursue it purely for the usability aspect. Kind of annoying to get zipped files then unzip them then process then you have extra unzipped files, plus in mass production, this may save some copying issues but will need to see. 

User-friendliness is a time improvement ;P

mts299, 2021-01-21:
Gonna close this for now and make a GitHub project for it.

closed 2021-01-21 by mts299