SuperDARN/rst #338: make_fit core dump

jaestephenson, 2020-09-17:
Hello
When I try to use the following code as an example: make_fit -old 2005050520d.dat 2005050520d.fit 2005050520d.inx
the program falls over and the core is dumped. It makes .fit and .inx files that don't work with old plotting software I wrote in windows, so clearly it does not make them correctly.
Is it possible that this could be rectified as I spent considerable time writing these programs and we still have lots of good events we find in our old data!
Thanks, Judy

ecbland, 2020-09-17:
Hi Judy

Thanks for letting us know about this. It looks like this problem was introduced in RST4.4, since the `make_fit -old` code works fine in RST4.3. 

By the way, we actually recommend converting `dat` files to `rawacf` format, and then running `make_fit` on that `rawacf` file. This is to circumvent the file compression in the old-format `fit` files.
```
dattorawacf 2005050520d.dat > 2005050520d.rawacf
make_fit 2005050520d.rawacf > 2005050520d.fitacf
```

@mts299 Would you mind taking a look at this?

mts299, 2020-09-17:
Sounds good, made another Issue if `-old` is still something we want to support.

mts299, 2020-09-17:
@ecbland ... so not sure if this a computer-dependent bug but on my machine, I have no issues :/ 
I cannot read the files with `dmapdump` but I am not sure if it can read `fit` and `inx` files. 

I am using `develop` so maybe it got solved already?

ecbland, 2020-09-17:
@mts299  I get the problem on `develop` also. The specific error is 
```
free(): double free detected in tcache 2
Aborted (core dumped)
```
`make_fit -old` works fine in RST4.3, so the error must have been introduced after that release. Could this be related to #292? I'm the person who tested that code, so perhaps I did not test it on old-format files 😮  

> @mts299 I cannot read the files with dmapdump but I am not sure if it can read fit and inx files.

`dmapdump` only works on the new format (`dmap`) files, so that's the expected behaviour. There are a few binaries for reading the old-format files, e.g. `make_info`

mts299, 2020-09-17:
huh I don't get that... must be with the new compiler or my compiler just handling it... I will try my laptop with Ubuntu on it.

mts299, 2020-09-17:
Soooo ubuntu shows no error either :/

mts299, 2020-09-17:
What OS are you using @ecbland and gcc compiler?

ecbland, 2020-09-17:
> What OS are you using @ecbland and gcc compiler?

Linux Mint 20 Cinnamon
gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0

If this is indeed related to #292, then I would have tested that PR on an older version of Mint (and probably gcc) that I no longer have access to.

mts299, 2020-09-17:
Do you have time to learn how to debug code through a chat and screen sharing?

mts299, 2020-09-17:
Does it give you a line number? I assume not.

jaestephenson, 2020-09-18:
I am using Ubuntu 16.04.7 LTS gcc/xenial, now 4:5.3.1-1ubuntu amd64
I noticed when I tried to use the same expression (make_fit -old.....) on sd-work9 at Virginia Tech, it gave the same problem.
Hope this helps!

egthomas, 2020-09-18:
> This is to circumvent the file compression in the old-format `fit` files.

@ecbland I do not think there is any additional compression / loss of information when going from `dat` to `fit` - I believe it is at the raw voltage sample to `dat` stage on-site that some precision is/was lost.

ecbland, 2020-09-18:
@egthomas Ok, thanks for the clarification. I may have misunderstood an earlier comment from [this issue](#236), where we determined that it's better to convert from `dat` files to `rawacf`, rather than running `make_fit -old` directly on the `dat` file.

egthomas, 2020-09-18:
@ecbland sorry, you are right!  I had forgotten about the skynoise issue you identified in #236, and from looking at the `fitwrite` code again, it appears all of the other fitted parameters are also compressed.  My mistake!

ecbland, 2020-09-24:
@jaestephenson Thanks for alerting us to this bug. We think we have fixed it now, so I will close this issue. If you'd like to test it yourself, you can download the [`develop`](https://github.com/SuperDARN/rst/archive/develop.zip) branch of RST and install it in the normal way.

We'll make a patch release that includes the bugfix ASAP.

closed 2020-09-24 by ecbland