SuperDARN/rst #339: make_fit -old deprecation?

mts299, 2020-09-17:
@ecbland and @ksterne, seeing how we are recommending people to convert dat files to rawacfs and then fitacfs due to compression issues in fit files, would it be worth wild to deprecate the -old option in maake_fit to avoid people using it? 

Maybe this is a DSWG question? 

Sorry, ignorant of the fit format. I have also had issues reading it in pyDARN and asked users to convert them to fitacf or if they have the dat files, then rawacf to fitacf.

egthomas, 2020-09-18:
I personally believe we should retain all of the `-old` library and binary functionality.

mts299, 2020-09-18:
Okay, I have no problem with this just noticed how we say one thing even though others try another way and maybe not the best analysis method... seems a bit bad to get legacy code that doesn't give the best results. 

@ecbland if you feel this shouldn't press on you are more than welcome to close this issue. This is just something I noticed and know from Software Developers point view legacy code is never health for software packages.

ecbland, 2020-09-19:
Always good to ask questions! Yes, this smells like a DSWG issue, but I think it's unlikely that the PIs would support it (in the near-future at least). I can certainly see your point about software "health" with legacy code, and it's helpful to recognize and talk about that. Annoyingly we are stuck with a legacy file format which we don't fully understand (e.g. https://github.com/SuperDARN/rst/issues/236#issuecomment-505049864), and is still widely used.

That being said, we should think about how to communicate our recommendation that users convert `dat->rawacf` rather than using `make_fit -old`. It's already in [readthedocs](https://radar-software-toolkit-rst.readthedocs.io/en/latest/user_guide/make_fit/), but that may not reach people who already know how to use `make_fit` :smile: 
Perhaps we could add an "RST important info" page in readthedocs that includes only the most essential information about data processing, and then promote it heavily to the community. This could include information about default processing options as well (e.g. default version of FITACF, grid/map defaults).

closed 2020-09-19 by ecbland

mts299, 2020-09-23:
submitted the issue to DSWG: https://github.com/SuperDARN/dswg-published-docs/issues/12