SuperDARN/rst #341: Patch release RST 4.4.2?

mts299, 2020-09-17:
Wondering if we want to do a patch release with 3 bug fixes on develop now that was brought to our attention by outside users. 

PR's to merge still:
- [ ] #340 
- [ ]  #313 
- [ ] #320?
- [ ] #337?  

I think these PR's are within a patch release definition and would be good to be out right away. 

I can test some tomorrow.

ecbland, 2020-09-17:
Agree with @mts299 that we should seriously consider a patch release, especially for #340 (fixes #338)

The bugfixes already on `develop` are:

- [ ] #331 `grid_plot` not plotting power or spectral width data in southern hemisphere (issue #330)
- [ ] #332 AACGM_v2 coefficients are not valid for altitudes above 2000 km (issue #332)
- [ ] #321 Incorrectly formatted `rawfp` structure in DLM version of `OldRawOpen`

ecbland, 2020-09-28:
@ksterne Any thoughts here?

ksterne, 2020-09-28:
I agree here. Sorry I've been behind on things. I haven't been in my office for over a week now, so a little disorganized.

I'll get a patch release branch going and cherry pick things to include those 4 PRs and adjust some other things as needed.

mts299, 2020-10-26:
Closing this since we agreed on a minor release.

closed 2020-10-26 by mts299

ksterne, 2020-10-26:
I think our discussions (maybe not as well on github) lead to at least keeping this open as a fall-back in case we didn't get to a minor release in a timely manor.  But it's probably good to leave it closed as I think we're working our way to a minor release.