SuperDARN/rst #344: Missing acfd array causing segfault in IDL

ecbland, 2020-10-05:
This issue is a follow up to #291, which fixed a segfault caused by a missing `acfd` array in a `rawacf` file. The problem was fixed in the C code (thanks @egthomas!), but it's still present in the IDL code. 

Using the same example as #291 for the DLM code:
```
IDL> unit=rawopen('20080103.2201.00.inv.rawacf')      
% Loaded DLM: RAWDLM.
IDL> s=rawread(unit,prm,raw)                          
IDL> while s ne -1 do s=rawread(unit,prm,raw)
Segmentation fault (core dumped)
```

And similarly with the native IDL code
```
IDL> unit=rawopen('20080103.2201.00.inv.rawacf',/read)
IDL> s=rawread(unit,prm,raw)
% Compiled module: REVERSE.
% Conflicting data structures: structure tag,<INT       Array[23, 2]>.
% Execution halted at: RADARDECODERADARPRM  275 /rst/idl/lib/superdarn/main.1.25/rprm.pro
%                      RAWREAD           118 /rst/idl/lib/superdarn/main.1.25/raw.pro
```

egthomas, 2020-10-05:
The DLM code was a pretty easy fix (https://github.com/SuperDARN/rst/commit/c56382751be390ce96b89bd1122de4bb70a34360) but it's not quite as obvious why the native IDL code is failing.  It may be something to do with the first record of that particular file, since when I tried reading a trimmed version of that RAWACF file with a different start time the native code worked fine.

ecbland, 2020-10-05:
Thanks @egthomas ! That commit fixes the problem for all of the files I looked at with the DLM code (`20161031.*.*.sto.rawacf`)

I think you're right that the native IDL code issue is more specific to that particular file from Inuvik. I can't reproduce the problem with the `sto` files (I hadn't checked that earlier).

pasha-ponomarenko, 2020-10-05:
Sorry for barging in: may it be that the lag array in one or more records have a dimension which is different from the default one, 23x2?

egthomas, 2020-10-05:
I think @pasha-ponomarenko is correct - the `ltab` array in that particular rawacf file (`20080103.2201.00.inv.rawacf`) has dimensions `[23][2]` according to `dmapdump`, when it should be `[mplgs+1][2]` (where in this case `mplgs = 23`), or `[24][2]`.

ecbland, 2020-10-25:
Closing this now that #346 has been merged

closed 2020-10-25 by ecbland