SuperDARN/rst #345: Bad phi0 calculation at DCE from 2018 onward

aburrell, 2020-10-05:
I have been looking into the DCE elevation angles recently, and found that a bug had been introduced into fitacf version 3.0 in the last release.  I didn't see any issues for this, but it's possible someone else has seen it too.  The problem is, all phi0 values are zero and they aren't if you run fitacf 2.5.  They also were not zero in the previous release of RST.  You can see where my processing was when the last release came out:

[image]

You can see the difference between running version 2.5 and 3.0 on the same rawacf file:

[image]

But this is not the case for all radars:

[image]

egthomas, 2020-10-05:
I noticed this behavior (phi0 = 0) for some recent BPK data processed using FITACF 3.0 as well, but I wasn't sure if it was a radar-specific issue or related to the FITACF 3 algorithm.

ecbland, 2020-10-05:
@aburrell Could you please clarify which two versions of RST you're comparing here?

aburrell, 2020-10-05:
1) The current master branch (v4.4.1) and the one right before it (v4.4) (I think!).  
2) Yes, it does apply to all DCE data (I went back and checked).  The date change just indicates that I updated RST at that point in the processing.

egthomas, 2020-10-05:
I thought the only change between v4.4 and v4.4.1 was to the stdkey makefile, which shouldn't affect any of the fitting algorithms, right?

mts299, 2020-10-05:
yea... I just changed a make thing which essentially going back

aburrell, 2020-10-05:
Maybe I had the version before that then... Sorry, I can see what I have now, but I am not sure how to see what I had before!

egthomas, 2020-10-05:
This is getting a little off-topic, but maybe we should think about appending the RST version number to the `origin.command` string since the FITACF major/minor revision numbers are not always sufficient to capture changes between releases?

mts299, 2020-10-05:
@egthomas can you make an issue for this comment so we can discuss it there and focus on the issue about the change in elevation here. I don't want to burry this comment. Gonna do some testing myself soon on this bug.

aburrell, 2020-10-06:
@egthomas did some work investigating the phi0 and elevation angles for different versions of RST and FitACF here: http://sdnet.thayer.dartmouth.edu/web_data/phi0_test/

After some discussion, we believe the issue in this case is that the hardware value for the phase sign (12) is zero at DCE and the elevation angle routine used by FitACF 3.0 expects a value of +/-1.  This is not an issue for FitACF 2.5 because the elevation angle routine there does not use this hardware value.

mts299, 2020-10-06:
Looks like another reason for Simon's code to get into FITACF 3.0

egthomas, 2020-10-06:
Note that FIR and DCN (and possibly other sites) also have a `phidiff` value of zero in their hardware files.  We have not yet been able to explain the odd `phi0` behavior at ADE or BPK.

ecbland, 2020-10-06:
That's some nice detective work @aburrell and @egthomas!

The radars with `phidiff=0` are: DCE, DCN, FIR, HAL

> @egthomas We have not yet been able to explain the odd phi0 behavior at ADE or BPK.

I'm a bit confused here. I can reproduce the problem for BPK on 20200909, but not for other dates, and not for ADE. The results that @egthomas posted on that link look ok for these radars as well.

egthomas, 2020-10-06:
> I'm a bit confused here. I can reproduce the problem for BPK on 20200909, but not for other dates, and not for ADE. The results that @egthomas posted on that link look ok for these radars as well.

Sorry, I should have clarified what I meant for those two sites.  For ADE the opposite is actually occuring - the FITACF 2.5 results show all-zero values of `phi0`, while FITACF 3.0 shows non-zero values of `phi0`.

For BPK, I was just referring to the above plot I shared; the all-zero `phi0` values on that date may be entirely unrelated to this `phidiff` issue afflicting the other radars.

ecbland, 2021-05-30:
This appears to be the source of the problem:

https://github.com/SuperDARN/rst/blob/97f26d65a45a83cc2f96c808682061911efa905b/codebase/superdarn/src.lib/tk/fitacf_v3.0/src/determinations.c#L725-L727

The `phidiff` multiplication was added in 2018 to obtain the correct phase sign for radars with `phidiff=-1` (see 7785a14). This will obviously cause problems if `phidiff=0`. I believe that any radar with an interferometer should have `phidiff=+/-1`. So, the problem is actually the hardware file.

-------------

I then dug into the `elevation_v2` code and noticed that Simon left some comments about `phidiff`. Note, however, that GBR is not the only radar with `phidiff=-1`. CVE, KOD, MCM, SPS also have `phidiff=-1` in their hardware files:
https://github.com/SuperDARN/rst/blob/97f26d65a45a83cc2f96c808682061911efa905b/codebase/superdarn/src.lib/tk/elevation.1.0/src/elevation_v2.c#L65-L89

### Action items (with PI consultation):

1.  Change `phidiff` to either +1 or -1 for all radars with an interferometer. **This will fix the bug reported here**
2. Find out what `phidiff` actually means and review its use in RST.

_Note: I have not investigated the other bugs reported here for BPK and ADE_

ecbland, 2021-06-15:
@aburrell I think we can close this now? The hardware file has been updated in the `hdw` repo (https://github.com/SuperDARN/hdw/commit/aab7b14c6bca46a4373b9b72a1abb79705cb6fca)

aburrell, 2021-06-15:
Yes, this addresses the issue.

closed 2021-06-15 by aburrell