SuperDARN/rst #347: Documentation of hardware files

aburrell, 2020-10-06:
The RST docs do not currently have a description of the hardware files.  This would be very useful.  The documentation provided by VTech would be a good start:  http://davit1.ece.vt.edu/hdw/hdw.dat_Description.pdf

Also, this documentation describes the value of the Phase Sign as being +/- 1.  This is not the case at all radars.  For one radar, FIR, this is likely because there is no interferometer.  If this is the accepted practice, it should be added to the description.

mts299, 2020-10-06:
I believe this is a DSWG thing and should host it with the new git repo. DAWG/RST could cite the documentation, but not sure we should write something up for it. 

Thoughts @ecbland

mts299, 2020-11-12:
@aburrell @ecbland Does the new hardware repo support the documentation you wanted? 
Shall we pass this to DSWG? 

DAWG could point to the repo + any new documentation from DSWG.

aburrell, 2020-11-12:
If you're referring to the https://github.com/SuperDARN/hdw README, it's a good start but not complete.  For instance, Phase Sign is allowed to be zero if there is no interferometer, and that isn't specified in the description.  It would also be useful to include the variable names these quantities are given in RST.  The RST documentation would also need to point to this repository, since it's not easy to find and would be even harder for someone new to the network.

mts299, 2020-11-12:
Okay, I think DSWG will need to hear what you want in the documentation since that is their responsibility. 

I will point to the repo and add which fields RST uses in our files.

mts299, 2020-11-12:
Side note: these are simple changes so this is something anyone can do if I can not get to it right away :p

mts299, 2021-01-20:
Was this solved with the hdw repo?

aburrell, 2021-01-21:
The core of the issue still exists.  I thought about transferring this issue to the hardware repo, but the RST docs should at a  minimum point to accessible (in the senses of being able to find it and understand it) and complete hardware documentation for this to be closed.

mts299, 2021-01-21:
@aburrell do you mind doing this since you found that pdf? 

I won't be able to touch it for a while now.

aburrell, 2021-01-21:
I don't have any time in my schedule for it at the moment.

ecbland, 2022-01-11:
This issue has gone stale so I'm closing it for now.

closed 2022-01-11 by ecbland