SuperDARN/rst #350: Unable to compile RST library

mtwalach, 2020-10-13:
Hi, 
I am still trying to install RST on a new OS (I have made an advancement on my last [issue ](https://github.com/SuperDARN/rst/issues/317) (🎉) but have now run into a problem that I have never come across before:

```
================================================================================
Compiling Library:/Users/maria/rst/rst/codebase/superdarn/src.lib/tk/sim_data.1.0
================================================================================
make clean
rm -f *.o
make
make.version /Users/maria/rst/rst/codebase/superdarn/src.lib/tk/sim_data.1.0/src/..
make.hdr /Users/maria/rst/rst/codebase/superdarn/src.lib/tk/sim_data.1.0/src/../include /Users/maria/rst/rst/include/superdarn 
/Users/maria/rst/rst/codebase/superdarn/src.lib/tk/sim_data.1.0/src/../include/sim_data.h
/Users/maria/rst/rst/codebase/superdarn/src.lib/tk/sim_data.1.0/src/../include/sim_data.h -> /Users/maria/rst/rst/include/superdarn/sim_data.h
cc -Wall -O3 -ansi -D_GNU_SOURCE -D_DARWIN -I/opt/local/include -I/Users/maria/rst/rst/include/base -I/Users/maria/rst/rst/include/general -I/Users/maria/rst/rst/include/superdarn -I/Users/maria/rst/rst/include/analysis   -c sim_data.c    
In file included from sim_data.c:37:
In file included from /Users/maria/rst/rst/include/superdarn/sim_data.h:5:
/Users/maria/rst/rst/include/base/rmath.h:44:1: error: declaration of anonymous struct must be a definition
struct complex {
^
/Users/maria/rst/rst/include/base/rmath.h:44:1: warning: declaration does not declare anything [-Wmissing-declarations]
1 warning and 1 error generated.
make: *** [sim_data.o] Error 1
Compilation Aborted.
```
I think this might be a problem in `/include/base/rmath.h` but I can't spot what is wrong with it - Can anyone help?

mtwalach, 2020-10-15:
Update: Apparently this is a known problem with the compiler on Catalina. I removed `make_sim.1.0`, `sim_data.1.0` and `sim_real.1.0` which all throw up this error and RST compiles fine without them. I have never used those routines before, so I won't miss them. If anyone does want to use these on Catalina however, they may run into this issue...

egthomas, 2020-10-15:
@mtwalach unfortunately I can't help with compiling on Mac / Catalina, but just wanted to mention that these same libraries / binaries gave us a lot of trouble several years ago:  https://github.com/vtsuperdarn/VTRST3.5/issues/19

At the time our solution was to tar the problematic directories so the `make.code` process would skip over them.  At some later point, I copied a few bug fixes from @asreimer's RSTLite fork which (at that time) got the `make.code` process working again (https://github.com/SuperDARN/rst/commit/d161d560a9463103efa78613625585f632d87dba).

asreimer, 2020-10-15:
Hey everyone. I don't have time to fix this, but I think the error message is pretty clear:

    /Users/maria/rst/rst/include/base/rmath.h:44:1: error: declaration of anonymous struct must be a definition
    struct complex {
    ^

The compiler doesn't like the anonymous structure definition. Some newer versions of compilers (remember all the extra warnings I found in RST with GCC 8? Latest is now GCC 9.2) do a better job of ensuring compliance with newer code standards and have started throwing errors where once they would just throw a warning.

You might be able to add `-Wno-gnu-anonymous-struct` to the makefile for the `sim_*` code as a bandaid work around.

The proper fix involves investigating where RST even uses that `struct complex` definition and just removing it and updating any code that might be using it to use the builtin "complex.h" instead. Custom type definitions from the time before the C99 Standard existed probably don't make sense 20 years later.

mts299, 2020-10-26:
@mtwalach did you fix this?

mts299, 2020-10-31:
Working on the solution @asreimer suggested :)

mtwalach, 2020-11-02:
Sorry, my previous post on this wasn't particularly clear. "fix" is a strong description: I took the aforementioned culprits of code out (i.e. removed the folders & files) and then it compiled just fine. As I said, I probably won't miss them but someone else might (they are neat pieces of code).

mtwalach, 2020-11-02:
Happy to test it when it's done!

closed 2020-11-17 by mtwalach