SuperDARN/rst #357: Elv_low Elv_high in Fitacf 3.0

mts299, 2020-10-29:
So I left the `elv_low` and `elv_high` the same as Pasha implemented them as (not the same as FITACF 2.5, @ecbland if you can explain better please do so!) in PR #356; however, because they have 2 meanings based on the algorithm picked and may not be aware to the user that the meanings have changed between versions. 

I am wondering if we want to convert FITACF 3.0 `elv_low` and `elv_high` back to the meaning of what FITACF 2.5 set it as and ask DSWG about creating 2 new FITACF fields for FITACFs 3.0 current `elv_low` and `elv_high`?

Please do not ask me specifics as I don't really know much except what is in the code.  However, suggestive ideas and solutions are most encouraged and what actions to take next on this given problem.

ecbland, 2021-05-18:
Re-opening this issue since we are waiting for PI input. (I think there was some confusion here with #410, which is already closed)

mts299, 2021-05-18:
Woops sorry closed the  wrong issue :p

ecbland, 2022-01-11:
Update: the PIs approved [our suggestion](https://superdarn.github.io/dawg/reports/202102/#fitacf3-elevation-fields) to add some new fields to the fitacf format at their meeting in November 2021. I already started the coding to implement this change...I'll make a PR when it's ready.

PR #472 (2022-10-11): Updated elevation angle field names to support FitACF 3.0
This PR introduces two new fields to the `fitacf` file format: `elv_fitted` and `elv_error` (see [February 2021 DAWG report](https://superdarn.github.io/dawg/reports/202102/#fitacf3-elevation-fields), closes #357). This change was approved by the PIEC in November 2021. 

Additionally, I decided to experiment by completely removing the XCF arrays that FitACF 3.0 does not populate (`x_v`, `x_p_l`, etc.) Previously these arrays were just filled with zeros, but now they are completely removed. **Please let me know if this is a good idea!**
 [... cut]
closed 2022-10-11 by egthomas