SuperDARN/rst #368: Compilation error on Ubuntu 20.04.1 LTS

horit, 2020-12-15:
Not quite sure this is the right place to report this kind of error, but let me describe a minor problem in compiling the RST. Recently I tried to compile the RST-4.4.1 (the official v.4.4.1 from the Zenodo site) on Ubuntu 20.04.1 LTS and encountered an error for which the primary part of its error message is copied below: 

```
================================================================================
Compiling Library:/home/g3/horit/local/sd/rst4/codebase/superdarn/src.lib/tk/idl/cnvmapidl.1.3
================================================================================
make clean
rm -f *.o
make
make.version /home/g3/horit/local/sd/rst-4.4.1/codebase/superdarn/src.lib/tk/idl/cnvmapidl.1.3/src/..
make.hdr /home/g3/horit/local/sd/rst-4.4.1/codebase/superdarn/src.lib/tk/idl/cnvmapidl.1.3/src/../include /home/g3/horit/local/sd/rst4/include/superdarn
/home/g3/horit/local/sd/rst-4.4.1/codebase/superdarn/src.lib/tk/idl/cnvmapidl.1.3/src/../include/cnvmapidl.h
'/home/g3/horit/local/sd/rst-4.4.1/codebase/superdarn/src.lib/tk/idl/cnvmapidl.1.3/src/../include/cnvmapidl.h' -> '/home/g3/horit/local/sd/rst4/include/superdarn/cnvmapidl.h'
gcc -I/usr/local/include -I/usr/local/include/ncurses -I/usr/include -L/usr/lib -L/usr/lib/x86_64-linux-gnu -fPIC -Wall -O3 -ansi -D_GNU_SOURCE -D_LINUX -I/home/g3/horit/local/sd/rst4/include/base -I/home/g3/horit/local/sd/rst4/include/general -I/home/g3/horit/local/sd/rst4/include/superdarn -I/usr/local/exelis/idl/external/include -c cnvmapidl.c
In file included from cnvmapidl.c:20:
/usr/local/exelis/idl/external/include/idl_export.h:994:1: error: C++ style comments are not allowed in ISO C90
  994 | // BOOLEAN: Check if var is a BOOLEAN
      | ^
/usr/local/exelis/idl/external/include/idl_export.h:994:1: note: (this will be reported only once per input file)
make: *** [/home/g3/horit/local/sd/rst4/build/make/makelib.linux:19: cnvmapidl.o] Error 1
Compilation Aborted.
```

Obviously the gcc with an option of "-ansi" does not allow C++ style comment lines in idl_export.h, causing the compilation error. I just removed "-ansi" from the CFLAGS in build/make/makebin.linux, makedlm.linux, and makelib.linux, and tried again, and apparently has compiled it successfully. I hope that this may be of help to somebody trying the same and that removal of "-ansi" would not have any side effect...

My pc environment:
gcc 9.3.0 and IDL 8.5.1 on Ubuntu Linux 20.04.1 LTS server

ecbland, 2020-12-15:
Hi @horit 
This is a known issue with the `idl_export.h` file that comes with IDL8.X, and yes it relates to the use of the C++ comments syntax. Only one line in that file is affected, so the easiest solution is to change the comments syntax from `//` to `/* ... */`:
```
// BOOLEAN: Check if var is a BOOLEAN
/* BOOLEAN: Check if var is a BOOLEAN */
```

We have actually removed the `-ansi` option for the next RST release (due this month), so this problem shouldn't come up again.

horit, 2020-12-15:
Thanks for the quick response and clarification! I am grad to hear that this error will no longer happen in the new release.

closed 2020-12-15 by ecbland