SuperDARN/rst #370: make_grid start time not working with concatenation option

mts299, 2021-01-08:
When I use:
`make_grid -st 04:04 -xtd -c 20100101*tig*fitacf > 20100101.tig.grid` 
The start record in the file is:
```
short   "start.hour" = 0
short   "start.minute" = 1
```

If I take `-c` option out: 
`make_grid -tl 60 -st 02:06 -xtd 20100101.0200.00.tig.fitacf > tmp.grd`
The start record is set correctly: 
```
short   "start.hour" = 2
short   "start.minute" = 6
```

egthomas, 2021-01-11:
Please try this fix and let me know if it works (`fix_make_grid_concat_bug` branch):

https://github.com/SuperDARN/rst/commit/0292a523c05590b58576c86ed5dfb574196efea0

mts299, 2021-01-11:
Hi Evan, Thank you for the effort in trying to fix this bug. However, I have already started working on this as I assigned this issue to myself.

mts299, 2021-01-14:
@egthomas if you would like to look at the PR in comparison to yours then that would be helpful.

closed 2021-05-18 by mts299